solvespace/solvespace #1402: Flathub version (installed via Discover) does not open in Manjaro (Arch)

mysigninname, 2023-09-03:
- SolveSpace version: 3.1 (released 2022.06.01)
- Operating system:** Manjaro 6.5.0-1

Expected behaviour:

Expecting the application to start

Actual behavior:

Application does not start

Additional information:

none at this time

dgramop, 2023-12-06:
Can you try invoking it from the terminal? Is there anything printed to the console?

mysigninname, 2023-12-06:
It says...
> Sorry, only UTF-8 locales are supported.

froqstar, 2024-04-26:
Same here

> ➜  ~ flatpak run com.solvespace.SolveSpace
Sorry, only UTF-8 locales are supported.
➜  ~ localectl                            
System Locale: LANG=en_US.UTF-8
               LC_NUMERIC=de_DE.UTF-8
               LC_TIME=de_DE.UTF-8
               LC_MONETARY=de_DE.UTF-8
               LC_PAPER=de_DE.UTF-8
               LC_MEASUREMENT=de_DE.UTF-8
    VC Keymap: us
   X11 Layout: us
    X11 Model: pc105
➜  ~

froqstar, 2024-04-26:
I remember from a past life that if I patch this locale check out and rebuild the flatpak myself, SolveSpace runs just fine.
What is this check supposed to prevent?