solvespace/solvespace #1403: Add solvespace to winget

Andrej730, 2023-09-04:
https://github.com/microsoft/winget-pkgs now seems to accept single-file software without msi installer. Adding solvespace would allow installing it along with the many other packages available on winget.

phkahler, 2023-09-04:
@Andrej730  there is no need to package solvespace for windows. As you point out it is a single executable. Someone is free to do that of course, but it's out of scope for upstream maintainers.

ruevs, 2023-09-05:
For cross reference https://github.com/solvespace/solvespace/issues/1309

ruevs, 2023-09-05:
There https://github.com/microsoft/winget-pkgs/pull/118952

ruevs, 2023-10-02:
Well it's been three weeks since I did (probably) everything needed to add SolveSpace to this "winget" tool... and still nothing...

It is pointless anyway... I doubt it that more than 10 people will ever use it to "install" SolveSpace.

phkahler, 2023-10-03:
Looks like it's in. Does this mean Windows will be able to find the app if someone tries to open a .slvs file? That would be really nice.

phkahler, 2023-10-03:
Searching at https://winstall.app/  finds it, but clicking on the site link takes me to our github and not solvespace.com.

Andrej730, 2023-10-03:
`winget install solvespace` works for me now.

ruevs, 2023-10-03:
Well the pull request got merged after I "complained" yesterday. https://github.com/microsoft/winget-pkgs/pull/118952

The link on https://winstall.app/apps/SolveSpace.SolveSpace points to the GitHub page inteed...

Here is the commit https://github.com/microsoft/winget-pkgs/commit/0e5bdb69309c432cb1e43c1893ba5663fdc14842 presumably it uses the "PackageUrl" and not the "PublisherUrl". Anyway...

closed 2023-10-03 by ruevs

Wallby, 2023-10-05:
> @Wallby should also " enjoy" this.

Nah, I think it's fine for people to have a different opinion than me. I'm glad though that it has been made a winget package as I think it helps reach a larger audience, even if it's not adding any functionality.