solvespace/solvespace #1405: posibility to export entity as ascii files

xyont, 2023-09-10:
Hi,

is there possible to export model entity to an ascii formatted text files e.g point, line and face constructor includes the index.

can someone give me such as guidance, where's the source files to modify? or the Developer also interested to implement for external CAE mesher (GPL).

Best,

ruevs, 2023-09-10:
https://solvespace.com/ref.pl#Export

SolveSpcae can export in the following formats:

Image:
	PNG image - png

2d view,
2d section: 
	PDF file - pdf
	Encapsulated PostScript - eps, ps
	Scalable Vector Graphics - svg
	STEP file - step", "stp
	DXF file (AutoCAD 2007) - dxf
	HPGL file - plt, hpgl
	G Code - ngc, txt

3d wireframe:
	STEP file - step, stp
	DXF file (AutoCAD 2007) - dxf

3d triangle mesh:
	STL mesh - stl
	Wavefront OBJ mesh - obj
	Three.js-compatible mesh, with viewer - html
	Three.js-compatible mesh, mesh only - js
	VRML text file - wrl

3d NURBS surfaces:
	STEP file - step, stp

Hopefully the above is enough ;-)

ruevs, 2023-09-10:
The exporting is done in the files starting with "export" here https://github.com/solvespace/solvespace/tree/master/src

closed 2023-09-10 by ruevs

xyont, 2023-09-10:
indeed, STEP format currently available are great features for data exchange of CAE/FEA mesher.  However, only tetrahedral  elemet of auto-meshing is possible and not hexahedral type. 

Previously i do some translation from [DXF](https://github.com/calculix/dxf2fbd) and [SKP](https://github.com/xyont/skp2fbd) to [FBD](http://www.dhondt.de/ov_prepo.htm), but it has many limitations. So, i interested in SolveSpace for the possibility. 

Thank you for pointing the links about exporting. source codes. I need to learn about structure data of SolveSpace workflow.