solvespace/solvespace #1407: CI, macOS: update action to use no instead of altool

ruevs, 2023-09-13:
### System information

- **[sign-macos.sh](https://github.com/solvespace/solvespace/blob/e6565c81186966a1c1bd7242e410419e02eb2774/.github/scripts/sign-macos.sh)**
- macOS GitHub CI

### Update needed to the macOS signing script

The GitHub  [action](https://github.com/solvespace/solvespace/actions/runs/6163878591/job/16729285064) building the "edge" version of SolveSpace in the "[Build Release macOS](https://github.com/solvespace/solvespace/actions/runs/6163878591/job/16729285064#logs)", on the "Sign Build" step gives a  [warning about `altool` being deprecated](https://github.com/solvespace/solvespace/actions/runs/6163878591/job/16729285064#step:5:288) and advises that a new "notarytool" should be used to sign the software.

```
2023-09-12 20:06:52.704 *** Warning: altool has been deprecated for notarization and starting in late 2023 will no longer be supported by the Apple notary service. You should start using notarytool to notarize your software. (-1030)
```

[sign-macos.sh](https://github.com/solvespace/solvespace/blob/master/.github/scripts/sign-macos.sh) should be updated to use this "notarytool ".

### Additional information

https://github.com/solvespace/solvespace/pull/1400#issuecomment-1716208224

ruevs, 2024-04-17:
https://github.com/solvespace/solvespace/pull/1408#issuecomment-2062106619

ruevs, 2024-04-17:
@vespakoen Naaah it did not work. And the build time became 24m55s (from 12m1s before) - it builds everything twice ;-(
https://github.com/solvespace/solvespace/actions/runs/8727893435/job/23946718599

vespakoen, 2024-04-17:
I see this:

Error: HTTP status code: 401. Invalid credentials. Username or password is incorrect. Use the app-specific password generated at appleid.apple.com. Ensure that all authentication arguments are correct.'

Maybe the password is not set correctly?

I can also generate a new one and send it to Jonathan so he can add it to the secrets.

I think it builds twice for different architectures? not sure why it's slower though.

I will try to find some time this week to look into it also look / try out the Qt stuff.

vespakoen, 2024-04-21:
Seemed to have been an issue with how it was retrieving the password from the environment variable. Fixed now...

ruevs, 2024-08-22:
@vespakoen @phkahler @jwesthues The CI on MacOS is failing again with:

> Error: HTTP status code: 403. A required agreement is missing or has expired. This request requires an in-effect agreement that has not been signed or has expired. Ensure your team has signed the necessary legal agreements and that they are not expired.'

https://github.com/solvespace/solvespace/actions/runs/10505848533/job/29104659047#step:5:73

Sorry for spamming all three of you but I do not know who "holds" the Apple "user" for SolveSpace.

vespakoen, 2024-08-22:
I just accepted the new agreement, hope that fixes it 😉

ruevs, 2024-08-23:
"It's alive!" thank you @vespakoen :-)
https://github.com/solvespace/solvespace/actions/runs/10526731759

ruevs, 2025-01-24:
@vespakoen I think Apple wants you accept their terms yet again :-)

https://github.com/solvespace/solvespace/actions/runs/12749119444/job/35531717656#step:5:73

vespakoen, 2025-01-24:
@ruevs correct, I accepted the new terms hope that does the job ;)

ruevs, 2025-01-24:
Works. @vespakoen thank you.

closed 2025-01-24 by ruevs