solvespace/solvespace #1416: Marquee selection similar to other CAD packages (left to right vs. right to left).

ruevs, 2023-10-14:
### System information

- **SolveSpace version:** https://github.com/solvespace/solvespace/commit/0d26ca17f750f62fee917a88101bdb03cf5225dd
- **Operating system:** all

### Expected behavior

[phkahler](https://github.com/phkahler) commented [Sep 24, 2023](https://github.com/solvespace/solvespace/issues/1374#issuecomment-1732564533)

> Some tools will select anything overlapping the box if you draw the box left-to-right and only things completely inside the box when drawn right-to-left.
>
> I thought solvespace was trying to do this but maybe not. Maybe it should. ;-)

[ruevs](https://github.com/ruevs) commented [Sep 25, 2023](https://github.com/solvespace/solvespace/issues/1374#issuecomment-1732668782) •

> Interesting! I never knew this existed - in any software.
> 
> This could be a pretty useful "power user" feature. I wonder how widespread it is? Apparently AutoCAD uses it https://forum.vectorworks.net/index.php?/topic/42974-marquee-select-like-autocad/ and it seems Adobe Illustrator does not :-) https://illustrator.uservoice.com/forums/333657-illustrator-desktop-feature-requests/suggestions/31863601-select-inside-marquee-window
> 
> And an example of the behaviour on the web
https://playground.babylonjs.com/#QHHXSD#1 (https://forum.babylonjs.com/t/marquee-selection-demo/16665)

### Actual behavior

The "direction" of marquee selection does not matter.

### Additional information

SolveSpace [marquee selection used to be "imprecise"](https://github.com/solvespace/solvespace/issues/1374) - any object whose Axis Aligned Bounding Box (AABB) intersected the marquee was selected. This was fixed for line segments in this [pull request](https://github.com/solvespace/solvespace/pull/1411) it is merged here https://github.com/solvespace/solvespace/commit/0d26ca17f750f62fee917a88101bdb03cf5225dd.

**Also possibly improve the performance of marquee selection on line segments: https://github.com/solvespace/solvespace/pull/1411#issuecomment-1733193151**

The marquee selection behaviour discussion started here https://github.com/solvespace/solvespace/issues/1374#issuecomment-1732564533 .

Not4sure, 2024-01-07:
This could also help users to overcome issues with current marquee selection behavior which still is "imprecise" for all except line segments. 
I made quick'n'dirty implementation to illustrate this.
[image]

I can make a PR but not sure if it would be better to solve current marquee issues first.

ruevs, 2024-01-07:
For this to work in a non-confusing manner it would be better if it found precise intersections with anything - arcs, circles, splines (text).

Bout you can always open a Work In Progress pull request if you want. This way your work will be visible and people can comment and maybe even contribute.