solvespace/solvespace #1418: Images from a sketch are "extruded" and get in the way

jkrei0, 2023-11-11:
### System information

- **SolveSpace version:** 3.1~36ecb85b 
- **Operating system:** Windows 10 

### Expected behavior

Images in a sketch stay in the sketch, and don't get duplicated when extruded.

### Actual behavior

Sketch images stay visible outside of the sketch, and are duplicated when extruding, making it hard to see anything (especially with an already busy model). Plus they cause z-fighting

[image]
[image]

### Additional information
I like to use images for reference and to trace organic shapes in sketches. Sometimes it's helpful to have these images show up outside of the sketch, but most of the time they just get in the way of things. I can just hide the sketch and extrude, but then I can't select any points or lines from them.

My preferred solution would be to have images be togglable (maybe with a "show images from this group" checkbox). Simply not having images show up in extrude groups is also a good solution, since I can then just hide the sketch and all the entities from it are still available from the extrude group.

phkahler, 2023-11-13:
@jkrei0  Not sure if you do code or build Solvespace yourself. I think the solution is to add one line to the code here:

https://github.com/solvespace/solvespace/blob/e7c0c1665f1684bb3195107147aaf254c852fa44/src/group.cpp#L1070C1-L1070C1

case Entity::Type::IMAGE:

right under the workplane one should exclude them from getting copied. I don't have any sketches using images, but I can make some and try this when I get around to it.  IMHO this seems like a good idea.

phkahler, 2023-11-14:
@jkrei0   I tried the fix proposed above and it works. It still makes extrusion lines from the 4 corners of the image, but not a copy of the image itself.  I made a pull request for the change but I try not to merge my own changes immediately.

ruevs, 2023-11-14:
@phkahler I think there is a better way. I wrote it today, but wanted to record a short video to explain. So I did not make a pull request. I'll do it tomorrow.

ruevs, 2023-11-15:
My idea is to not copy/extrude images only if they are marked "construction".
Here is a video that explains the idea:

https://github.com/solvespace/solvespace/assets/15338069/5354c156-f873-48ee-8d33-730dad260611

And here is a pull request https://github.com/solvespace/solvespace/pull/1420

PR #1420 (2023-11-22): Do not include construction images in extrude/lathe/revolve/helix
To avoid unnecessary/annoying copies of images added in 2d sketches if they are marked "construction" they will not be copied when creating solids.

Fixes: #1418
commit 70382660 (2023-11-22): Do not include construction images in extrude/lathe/revolve/helix

To avoid unnecessary/annoying copies of images added in 2d sketches if they
are marked "construction" they will not be copied when creating solids.

Fixes: #1418
closed 2023-11-22 by phkahler