solvespace/solvespace #1423: Freeze and crash on saving images

jkrei0, 2023-11-17:
### System information

- **SolveSpace version:** 3.1~0fd3af58 (latest, plus some fixes in my fork)

- **Operating system:** Windows 10 Pro 19045

### Expected behavior
File saves as normal

### Actual behavior
When attempting to save files with (some) images, SolveSpace freezes and cpu usage spikes. If you let it run for many hours, it'll eventually crash.

### Additional information
It seems to be dependent on the image file (crash happens only with some image files, but consistently), although I can't tell what the pattern is. Filename, path, drive, etc all seem to make no difference.

It also doesn't seem to change anything if there are other things in the file, or if the file has been saved previously.

Here's some example images and save files:
[test_crash_save.zip](https://github.com/solvespace/solvespace/files/13398196/test_crash_save.zip)

jkrei0, 2023-11-18:
I did some more testing, and it seems I was wrong, and it does have something to do with file location. If the image is not on the same drive as the slvs file, it fails, and the specific image has nothing to do with it. (i.e image from `C:\Users\User\Downloads`, but .slvs file saved to `H:\projects\solvespace\` causes the crash.)

I also tested it with an older solvespace build (`3.1~36ecb85b` w/ AMD fix), and it had the same issue.

ruevs, 2023-11-23:
It turns out images on a different drive are currently not supported. Here is what happens:

First it tries to make the path to the linked image relative in this context:

```
>	solvespace.exe!SolveSpace::Platform::Path::RelativeTo(const SolveSpace::Platform::Path & base) Line 366	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::SaveUsingTable(const SolveSpace::Platform::Path & filename, int type) Line 249	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::SaveToFile(const SolveSpace::Platform::Path & filename) Line 319	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::GetFilenameAndSave(bool saveAs) Line 588	C++
 	solvespace.exe!SolveSpace::SolveSpaceUI::MenuFile(SolveSpace::Command id) Line 701	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::WindowImplWin32::WndProc(HWND__ * h, unsigned int msg, unsigned __int64 wParam, __int64 lParam) Line 1064	C++
 	[External Code]	
 	solvespace.exe!SolveSpace::Platform::RunGui() Line 1712	C++
 	solvespace.exe!main(int argc, char * * argv) Line 29	C++
 	solvespace.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 41	C++
```

Since the image is on a different drive this fails and returns an empty path here:
https://github.com/solvespace/solvespace/blob/727b0153f6d07f9031862a4e2ecb0f4d30e87b8e/src/platform/platform.cpp#L366

And then it should (intentionally) crash here:
https://github.com/solvespace/solvespace/blob/727b0153f6d07f9031862a4e2ecb0f4d30e87b8e/src/file.cpp#L250

With this message:
[image]

@jkrei0 doesn't it do that for you?

jkrei0, 2023-11-23:
It does eventually give me that dialog, if I let it sit for a while (not sure how long, it took letting it sit overnight). But until then, it just seems frozen - with a spinny mouse cursor, and cpu usage goes up to a mostly consistent 20% (for reference, when saving a file "normally", it spikes to ~ 20% cpu for only a second).

And windows never complains about it like it usually does for unresponsive programs.

I'll try letting it go for a while again and see if I can get it to generate a debug report.

ruevs, 2023-11-24:
It does the same for me if I run it "alone" (not debugging from Visual Studio).

Try `ALT+Tab` it seems the erre message box does not come to the front somehow (x64 Windows 10 22H2 Version 19045.3636) ... 
It used to... some Windows update changed something perhaps?

jkrei0, 2023-11-24:
I've left it running for well over 24 hrs now, and it's still frozen and stuck at a consistent 20% cpu usage. It's totally frozen now, the window title/borders have disappeared, I can't minimize it, and the mouse cursor doesn't do anything (it doesn't even show as "spinning"). I'd rather not have my computer sitting there running it all the time, so I've given up and force-killed it.

Not sure why I got an error message before, and not now.

ruevs, 2023-11-25:
No need to generate a debug report - I debugged it - and it is perfectly clear what happens.

Linked files on different drives are simply not supported right now because SolveSpace wants to store a relative path in the .slvs file. This applies only to Windows since Unix-es do not have drive letters so all paths can be relative.

Whether this should be supported or not is not clear. The advantage of relative paths is that if the top level directory of a complex project with many linked files (in sub directories) is moved to a different location it everything continues to work.

The disadvantage is obviously this issue.

Even if we decide not to support absolute paths on Windows (that would be my preference) I think it should not be a crash but a "civilized" error message with an explanation. But this will take some doing.