solvespace/solvespace #1427: "Bad selection" should tell user what have they selected wrong

ror6ax, 2023-11-26:
### System information

- **SolveSpace version:**  3.1-flatpak
- **Operating system:** Fedora 38

### Expected behavior

On selecting improper elements to perform operation on, Solvespace gives feedback as to what did user do wrong.

Example: 
```
Bad selection for lathe group 

You have selected:

* some wrong types of objects
* and 
* some ok things

Lathing of <some wrong types of objects> is not possible.

<rest of the current message like "this group can be created with:">
```

### Actual behavior

On selecting improper elements to perform operation on, SolveSpace enumerates all eligible types of elements that could work, but does not tell the user what were they doing, thus making it hard for new users to learn what they need to change. 

[image]

phkahler, 2023-11-27:
The property browser will tell you what is selected, so presuming the user doesn't understand the terminology they can learn what things are called there.

The error messages presume you want to use a tool and maybe aren't sure what needs to be selected (axis for lathe for example) or what the tool can be applied to (there are some less obvious options)

So you'd want the massage to say what it does but also "invalid selection: point and line" for example?

BTW that's a more dynamic message than what we have.

ruevs, 2023-11-27:
Imagine the user has a sketch with hundreds of elements. After pressesing `CTRL-A` (select all) he tries to do a lathe. In this case what you are asking for would list hundreds of elements in the error message - this is neither helpful nor sensible. In addition - as @phkahler explained - the currently selected elements (if they are reasonably few) are listed in the Property Browser together with many helpful details for many of the combinations.

closed 2023-11-27 by ruevs