solvespace/solvespace #1428: If I write any uppercase letter (for example R15) into dimension input box solvespace crash

dadoprom, 2023-11-29:
### System information

- **SolveSpace version:** 3.1~727b0153 - Latest from git
- **Operating system:** Lubuntu 22.04.3

### Expected behavior

I sketched a circle and wanted to add dimension as radius with R15 in the dimesion input box, this is not working

### Actual behavior

If I write any uppercase letter (for example R15) into dimension input box solvespace crash

### Additional information

Solvespace does not crash when you write lowercase letters into dimension input box

phkahler, 2023-11-29:
@dadoprom  This is a known issue and will be fixed when we allow more arbitrary input for #77 Parametric sketches.

If you want to enter a radius instead of a diameter, click on the dimension (just select it, don't double click to edit. The dimension will show in the text window and a checkbox will allow you to change it from diameter to radius.

dgramop, 2023-12-06:
Right now solve space does crash (like the process dies) when you do this. I was able to replicate

If your dimension starts with a letter, then it crashes. Otherwise it gives you the popup as expected

Action:
Type in `R15` into a dimension box.

Actual behavior: 
Solvespace crashes

Expected behavior: A popup telling you you can't put a letter there
[image]