solvespace/solvespace #1431: slvs c++ library dosent constrain with SLVS_C_PT_PLANE_DISTANCE

Vishnu-C, 2023-12-05:
### System information

- **SolveSpace version:** 3.1~727b0153 
- **Operating system:** Windows 10

### Expected behavior

SLVS_C_PT_PLANE_DISTANCE should bring the point 3D above the work plane. [doc](https://github.com/solvespace/solvespace/blob/94618d4ce14a984a5a02398ace853828bb32cda6/exposed/DOC.txt#L280)

### Actual behavior

The point 3D remains unchanged

### Additional information

#### Code snippet

```
Slvs_hGroup g = 1;
sys.param[sys.params++] = Slvs_MakeParam(1, g, 15.0);
sys.param[sys.params++] = Slvs_MakeParam(2, g, 0.0);
sys.param[sys.params++] = Slvs_MakeParam(3, g, 0.0);
sys.entity[sys.entities++] = Slvs_MakePoint3d(101, g, 1, 2, 3);

std::vector<double> U = {1.0, 0.0, 0.0};
std::vector<double> V = { 0.0, 1.0, 0.0 };
double qw, qx, qy, qz;
Slvs_MakeQuaternion(U[0], U[1], U[2], V[0], V[1], V[2], &qw, &qx, &qy, &qz);

sys.param[sys.params++] = Slvs_MakeParam(41, g, qw);
sys.param[sys.params++] = Slvs_MakeParam(42, g, qx);
sys.param[sys.params++] = Slvs_MakeParam(43, g, qy);
sys.param[sys.params++] = Slvs_MakeParam(44, g, qz);

sys.entity[sys.entities++] = Slvs_MakeNormal3d(301, g, 41, 42, 43, 44);

sys.entity[sys.entities++] = Slvs_MakeWorkplane(401, g, aCuboid1->positionParamIdx, 301);

sys.constraint[sys.constraints++] = Slvs_MakeConstraint(
   ++constrainIdx, g,
   SLVS_C_PT_PLANE_DISTANCE,
   SLVS_FREE_IN_3D,
   fDist,
   101, 0, 401, 0);
   
Slvs_Solve(&sys, g);
```

phkahler, 2023-12-29:
@Vishnu-C  can you verify that the plane didn't move? Also, can you start with the initial point not on the plane?

Vishnu-C, 2023-12-30:
@phkahler The plane did move, but I solved it anyway, I had to  make the test point in a new separate group and solve with the new group. Thanks for the reply!

phkahler, 2023-12-31:
@Vishnu-C you could also constrain the plane so it can't move. But your approach of making a new group has the same effect since previous groups are held fixed while the later one is solved.

Can we close this issue?

Vishnu-C, 2024-01-01:
@phkahler I couldn't constrain the plane, can you suggest please? 
I believe my approach is a workaround, I may stumble upon cases where I cannot create a separate group for work plane.