solvespace/solvespace #1440: Ways to improve SolveSpace constraint Solver

Evil-Spirit, 2023-12-27:
Many years ago I was working on CAD based on SolveSpace, then SolveSpace itself, then NoteCAD - solve-space-like software built from scratch using clean-room-design, and finally now I am working on brand-now commercial propreitary solver, I can say that SolveSpace solver is very good in ideas, but having some problems which I already solved during a lot of job done after this.
Now you can see what can be done with the SolveSpace solver.
The project of C++ Solver inspired by SolveSpace and my job based NoteCAD called [Adjacent](https://github.com/Evil-Spirit/Adjacent)
This is NoteCAD solver ported by @wolfv from C# to C++.

phkahler, 2023-12-27:
@Evil-Spirit  What is the license for Adjacent? The code is publicly visible. If you have not chosen a license yet, I would recommend "GPLv2 or later" if you want to profit from it, and add MIT as a second option if you don't want to make money ;-)  Just my opinion of course.

Evil-Spirit, 2024-01-15:
Fixed.

xerio123, 2025-05-03:
sorry but, is adjacent support spline constrain, I mean is the solver can handle spline solved about a point ?