solvespace/solvespace #1441: Extrusion that includes Bezier spline behaves strangely

iyesin, 2024-01-01:
### System information

- **SolveSpace version:** `3.1~flatpak`
- **Operating system:** Debian 12 (bookworm)

### Expected behavior

Extrusion do a nice cut without removing existing surfaces.

### Actual behavior

Not sure how to describe it in words (I'm a newbie to CADs). Basically, when I'm trying to make a cut in a box with Beizier spline, at least one surface disappears from the resulting figure. Replacing a spline with an arc segment works nicely with a same box. See project and video attached.

### Additional information

https://github.com/solvespace/solvespace/assets/13554666/b1619095-8309-441b-b593-b7413b7387f9

[solvespace_Bezier_spline_extrude.zip](https://github.com/solvespace/solvespace/files/13805587/solvespace_Bezier_spline_extrude.zip)

phkahler, 2024-01-02:
@iyesin  I believe this is a duplicate of #1291 The NURBS boolean operations fail when there are 3 surfaces all tangent at the edge of a cut or intersection. There are some work arounds:

1) you could do the entire part as a single extrusion. This will not require any boolean combination of different extrudes.
2) Make the initial extrusion just a little longer than needed. Then cut away the entire end. This prevents too many coincident surfaces.
3) check "force NURBS surfaces to triangle mesh" in the group where the failure occurs. This will make it impossible to export the model as STEP file, and will have negative impact on performance if you keep adding more to the sketch. So not a preferred solution but it will let you model it exactly as-is.

iyesin, 2024-01-03:
Thank you for a quick response, @phkahler!

1. Workaround number one is not for me, because I want a curly cut in somewhat complex part.
2. I'll try workaround number two. And let me validate if get you right: I will need to create same part with Beizier spline cut but closed one. Then, after 1 extrusion, I cut excess off. Is this what you mean?
3. I did exactly that after all, actually, but I wanted to make it true STEP export.

iyesin, 2024-01-03:
@phkahler just for the note: I tried workaround number 2, but it gave me same result after cutting the excess.
[image]

phkahler, 2024-01-04:
@iyesin  I meant to cut the curve and the excess with a single extrusion. Draw the curved part as one side of the rectangle that cuts the excess.

phkahler, 2024-01-04:
@iyesin this is what I meant by number 1, doing it as a single extrusion:

[image]

The slvs is here:
[spline_fixed.zip](https://github.com/solvespace/solvespace/files/13832580/spline_fixed.zip)

iyesin, 2024-01-04:
@phkahler Oh. Now I get it. Still, this solves an issue only partially because I need this cut to be a part of a bit more complex assembly.

closed 2025-05-16 by phkahler