solvespace/solvespace #1448: Use 3rd degree NURBS surfaces for helix extrusions

phkahler, 2024-02-08:
### System information

- **SolveSpace version:** All
- **Operating system:** All

### Expected behavior

Helical surfaces should be mathematically correct - or as close as possible to correct.

### Actual behavior

We use 2nd order curves to form the edges of helical extrusions. This leads to non-uniform pitch along the extrusion. This error would probably be reduced by using NURBS of degree 3 enough that we'd never need to worry about it again.  One comment on this is here: https://github.com/solvespace/solvespace/issues/1447#issuecomment-1934270220

### Additional information

The function for creating these surfaces is here:
https://github.com/solvespace/solvespace/blob/7d379e7f3b045ae579537abd392d5b368a4117e5/src/srf/surface.cpp#L68

This function was modified for version 3.0 but is equivalent. An alternative would need to produce curves with 4 control points instead of 3, and the associated weights. I'm not currently up on the math for that. One could use degree elevation to go from 2nd degree to 3rd in the plane perpendicular to the axis, but the axial displacements will still need figuring out.

We probably want to keep the existing 2nd degree surfaces for revolve because there is no error for that case.

Evil-Spirit, 2024-05-21:
You should be imformed that the helical surface can't be repesented nor 2nd, nor 3rd order surfaces. So this is only about lowering number of subdivisions (read as performance).

phkahler, 2026-11-03:
@alken1t15   It is still open.  I will say that Evil-Spirit is right and there is no way to perfectly match such a surface with rational beziers of any degree. It's true that we could do better just by increasing the number of NURBS we use - just decreasing the angle reduces the error.  But why not use higher degree if it can help? Can it? I don't know ;-)

BTW if you make a helix of a few turns, set it to fixed pitch and slide the length or angle very slowly you can sometimes see the surface wobble a little bit as the patches sweep angle changes.

Evil-Spirit, 2026-11-03:
> But why not use higher degree if it can help? Can it? I don't know ;-)

Sinusoidal wave can't be expressed with polynom of finite degree. Helix projection having sinosuidal projection to axis. Now we have AI, why to use NURBS, may be we can create kernel which supports arbitrary equation shapes?

phkahler, 2026-11-03:
AI tells me the way to go is to use degree elevation to take our 2nd degree NURBS arc, elevate it to 3rd degree which is a straight formula to compute 4 points from 3 (the two ends stay the same though) and offset the 2 middle ones axially via some other formula. For a NURBS arc of some angle displaced into a helix of height H, the maximum percent axial deviation for 2nd and 3rd degree are:

[image]

We allow curve segments up to 90 degrees, so it's pretty bad at degree 2 getting H * 0.015 deviation.

@Evil-Spirit  You're not wrong. Parasolid can represent helical sweeps exactly and a few other mathematical shapes. Other than that I don't think its a good idea to roll our own non-standard surface representation.

alken1t15, 2026-11-04:
@phkahler Thank you for the additional analysis. I'd like to agree on a concrete scope before starting implementation. Is anyone already working on this?

I looked at `SSurface::FromRevolutionOf`, which currently uses degree 2 for both revolutions and helical sweeps. Would the following initial plan be useful?

1. Reproduce the pitch variation and build a small numerical comparison against an analytic helix, before changing the kernel.
2. Evaluate degree elevation of the rational arc to degree 3, with axial offsets for the two interior control points. Check maximum axial error, radial error, and endpoint/tangent continuity between patches over different sweep angles and pitches, including reverse sweeps and small angles.
3. If the measurements support it, apply the change to helical sweeps while preserving ordinary revolutions, then add regression coverage for representative models and downstream operations such as booleans/export.

Could you share the axial-offset formula or calculation behind your graph? What error reduction and continuity conditions would you consider sufficient for a useful first PR? Would you prefer comparing the cubic approach with smaller quadratic patches before choosing the implementation?

If this remains too exploratory, are there other current, unclaimed geometry or rendering issues that you would particularly like a contributor to tackle? My priority is a well-defined, useful contribution, and I'm happy to agree on a smaller scope first.

phkahler, 2026-11-04:
@alken1t15  Haha that function was my first significant contribution to the code!  Your update can probably be completely contained in that function.  You can see that for each point on a given input curve it generates a row of control points for the output surface. I would change it to have a local variable for that row of points - an SBezier - because each point is swept into a curve, or at least that's how I think of it. Calculate 2nd degree points for that sweep curve the same way they are done now. Then offset those curve points the same way they are now, and copy the curve points into the surface at the end of the loop. This should work exactly like it does now.

Then add code so that IIF there is axial displacement (this is a helix) do the degree elevation for the curve in-place. Copy the 3rd control point to the 4th. Then calculate the new 3rd and then 2nd points. Then apply the axial displacements to those points and let the curve be copied to the surface same as above.

I don't know what kind of testing to do for this. I don't want tests in the repository, just the resulting code in this function. I suppose a document (do we have dev docs folder?) showing the analysis of axial surface deviation before and after the change would actually be pretty cool.

I don't know the math behind the plot a posted, just standard Google gave me that after discussing it. The math to go from a 2nd degree Rational Bezier to 3rd is standard and can be found online. Not sure about how to axially display the 2 middle control points. Google gave a formula but I don't have it nor know if it was correct ;-)

phkahler, 2026-11-08:
@alken1t15  good plan. Go for it.

phkahler, 2026-11-08:
@alken1t15   There are several issue tagged help wanted and maybe a couple tagged good first issue:

https://github.com/solvespace/solvespace/issues?q=is%3Aissue+state%3Aopen+label%3A%22good+first+issue%22%2C%22help+wanted%22

I can take care of this one if you haven't started it yet.

alken1t15, 2026-11-08:
@phkahler Thanks! I'm working on this now. The implementation and validation are in progress; I'll share the results and a PR once they're ready.

alken1t15, 2026-11-08:
@phkahler The implementation is ready for review in #1813. Cubic helical patches preserve the circular projection and reduce the sampled axial error at 90 degrees from about 2.11% to 0.0493% of patch height. Ordinary revolutions remain quadratic. All 306 tests pass, including the new helix, shell, STEP and Boolean checks. Thanks for the guidance!

closed 2026-11-10 by phkahler