solvespace/solvespace #1449: Solver fails on constrain that just fixes current orientation if added after importing another object

zwn, 2024-02-09:
I made a comment 2 days ago https://github.com/solvespace/solvespace/issues/1247#issuecomment-1932818087 but it seems it might be unrelated/different problem, so here comes separate issue.

> I have also a somewhat simple reproduction case:
> - create a block (extruded rectangle) and save file
> - create a new file to be used as assembly
> - imported the block into the assembly
> - constrain the orientation of the imported block (_constrain normals in same orientation_) on the Z axes - works as expected
> - undo the constrain
> - import the same block for a second time
> - try to add the same constrain that worked before on the firs block - unsolvable
> - try to add the same constrain on the newly imported block - works as expected

[unsolvable.zip](https://github.com/solvespace/solvespace/files/14201184/unsolvable.zip)

I tried also with 2 different objects - block and cylinder.

### System information

- **SolveSpace version:** 3.1-7d379e7f3 self-built and current 3.1 release and recently nightly
- **Operating system:** Debian 12 and Windows

### Expected behavior

When the constraint can be added on the object before importing/linking another object,
it should be possible also after.

### Actual behavior

Solver fails with unsatisfied constrain referring to the same orientation constrain.

zwn, 2024-02-09:
I've built current master on Debian to check what is going on. It fails here:

https://github.com/solvespace/solvespace/blob/7d379e7f3b045ae579537abd392d5b368a4117e5/src/system.cpp#L326-L328

The actual value is equal to 1.0 and never changes (for `i == 1`).

The minimal reproduction case is:
- start solvespace and "save as" current empty view as `empty.slvs`
- file -> new
- import `empty.slvs`
- import `empty.slvs` for a second time
- try to constrain the Z axis of the first import to the global Z axis - fail
- try to constrain the Z axis of the second import to the global Z axis - success

It would be great if anyone with some knowledge of the related code could give me any hint where to go next. I am willing to get it down to a PR. I believe it must be something really simple at the end but fixing it would help me tremendously.

BKLronin, 2024-02-09:
If you select the previously imported model in the listview on the right it should work. It's not selected as current model so the solver fails.

From memory at least.

zwn, 2024-02-09:
Indeed, if I make the first imported object "active" in the list view first and only then add the constraint, the solver does succeed!

That was unexpected. Up until now I thought I grok _the solvespace way_ so this caught me somewhat off-guard. In my mind the imported objects were separate entities so it didn't hit me that I should operate only on the last one. It exactly mirrors the behavior when modeling a single object - when I create 2 sketches and try to add a constraint to the first one with the second one active, the solver fails as well. But my mental model was more like being in a sketch where it does not matter which point is added first.

So... It shouldn't allow me to add these constraints at all? Like some check when adding the constraint that at least one of the entities is in the active group?

BTW: I just love the software. Big thanks to everyone involved. It can do so much for such a tiny binary.

BKLronin, 2024-02-09:
Cool,

I do see it a bit as some sort of "timeline" with a single object in focus. Kind of. So when I add any external parts to check if things fit I can go back in time edit the base part then jump forward in time to check if the parts still fit etc. That way I can export a part without imported parts as well. (Random thought)

Same, I think SolveSpace feels kind of like a natural extension of thoughts on a project. Low on distraction and very context sensitive instead of 100 buttons. Sure you have to draw more in comparison but changing values is super solid even through more complex parts.

phkahler, 2024-02-09:
> If you select the previously imported model in the listview on the right it should work. It's not selected as current model so the solver fails.

Oh that's expected behavior. No bug here.

Our friend Evil-Spirit suggested adding a checkbox to each group to "Solve with previous" which would combine the current and previous group for solving together. This could be chained for several groups. It's a simple extension that would allow solving an entire assembly at once. Not sure how hard that would be to implement but I imagine not too hard once we figure it out. Also not sure that's the approach we want to take.

BKLronin, 2024-02-09:
Would that even be necessary? Usually objects are imported gradually and the last part is usually constrained to somehting that was there before. 

Step by step.

zwn, 2024-02-09:
> > If you select the previously imported model in the listview on the right it should work. It's not selected as current model so the solver fails.
> 
> Oh that's expected behavior. No bug here.

I was really confused by the behavior. If the added constraint is not going to be part of the solving process, I would expect to get some kind of complain (something like "don't create constraint outside of the active group"). Just saying "solve failed! unsolvable constraints" for something that is definitely solvable was confusing for me.

ruevs, 2024-02-12:
To summarize:

- Constraints created in the current group can "move"/modify only entities from the current group.
- They can **_not_** "move"/modify entities from previous groups.
- They **_can be_** against entities in any previous groups or the current one.

All of the above is by design.

The failure when a constraint is trying to "move"/modify entities from previous groups is confusing - the screen becoming red and the message `SOLVE FAILED! unsolvable constraints`.
[image]

We should try to come up with a better error message but (without looking at the code right now) it may not be easy to distinguish this case from truly unsolvable constraints.

ruevs, 2024-08-17:
As is usual with SolveSpace - it turns out that we already have an issue for this :-)
https://github.com/solvespace/solvespace/issues/236

closed 2024-08-17 by ruevs