solvespace/solvespace #1450: Add a clamped distance constraint

dcousens, 2024-02-11:
This is a feature request/discussion.

Instead of defining an exact distance value as a constraint, you could define the `minimum` and `maximum` distances, thereby ensuring the constraint prevents a segment, but leaves the design as parametric or [reasonably] unconstrained.

ruevs, 2024-02-11:
This is a duplicate to, or covered by:
https://github.com/solvespace/solvespace/issues/81
https://github.com/solvespace/solvespace/issues/77
https://github.com/solvespace/solvespace/issues/76
https://github.com/solvespace/solvespace/issues/422

It is the first point in the pinned issue above:
https://github.com/solvespace/solvespace/issues/1024

dcousens, 2024-02-11:
Thanks @ruevs, and sorry for missing that

closed 2024-02-11 by dcousens

ruevs, 2024-02-11:
No problem :-)