solvespace/solvespace #1452: Naked edges with only rectangles and extrusion

kristianlm, 2024-03-07:
Hi, and thanks for this great "little" piece of software with so much punch. SolveSpace is the first and only CAD I've learned to use.

I've got a situation that requires me to force everything to triangle meshes which doesn't seem right. Every object is 90 degrees as I'm only using rectangles. I've managed to reproduce this on the Arch Linux AUR package and on my GNU Guix system. I don't think it's a packaging issue.

### System information

- **SolveSpace version**: 3.1~70bde63c
- **Operating system:** Reproduced on Arch Linux and GNU Guix

### Expected behavior

[image]

I've forced `NURBS to triangle mesh` is on to get that screenshot

### Actual behavior

[image]

When `Force NURBS to triangle mesh` off, I'm missing an entire side surface.

### Additional information

[bug-report-naked-edges-with-only-cuboids-extrude-and-unions.zip](https://github.com/solvespace/solvespace/files/14521863/bug-report-naked-edges-with-only-cuboids-extrude-and-unions.zip)

The model is easily produced as follows:

[image]

- Make rectangle A
- Extrude it downwards
- Make rectangle B in new sketch using previous sketch workplane or P
- Extrude it upwards
- Make rectangle C in new sketch using B's equivalent of point P (selecting its two connected lines)
- Extrude it upwards
- Make rectangle D in new sketch using P and its two perpendicular lines (or point on the other two cuboids)
- Extrude it right

The extrusion on A is in the opposite direction from B's extrusion. I cannot reproduce this bug if both are extruded in the same direction. The bug depends on the height of D: If D touches just A and B everything is ok. What's very strange is that if D if just _a little bit_ above C, it still works:
[image]
Here, D touches C just a tiny bit and `force NURBS to triangle mesh` is off!

phkahler, 2024-03-12:
@kristianlm can you provide a simple .slvs file demonstrating this?

kristianlm, 2024-03-12:
> @kristianlm can you provide a simple .slvs file demonstrating this?

Indeed, it is there in my original post, though perhaps not as clearly marked as it could have been.

ruevs, 2024-03-13:
Thank you for the nice example. It falls in the second group here https://github.com/solvespace/solvespace/issues/738, I've added it.