solvespace/solvespace #1454: glCreateShader fails on MacOS M1

dgramop, 2024-03-24:
### System information
MacOS M1 on 87d0e097bbbfb895e622cf970f236bcb108167f6

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior
Solvespace should launch, in particular, glCreateShader should return a non-zero value

### Actual behavior
Solvespace crashes. Running in lldb, and inspecting the stack after crash:

TL;DR assertion fails when trying to create shaders/edge.vert
```
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x000000018b81e0dc libsystem_kernel.dylib`__pthread_kill + 8
    frame #1: 0x000000018b855cc0 libsystem_pthread.dylib`pthread_kill + 288
    frame #2: 0x000000018b761a40 libsystem_c.dylib`abort + 180
    frame #3: 0x0000000100059ee8 SolveSpace`SolveSpace::Platform::FatalError(message="File /Users/dgramop/Projects/solvespace/src/render/gl3shader.cpp, line 104
, function CompileShader:\nAssertion failed: shader != 0.\nMessage: glCreateShader failed.\n") at guimac.mm:90:5
    frame #4: 0x000000010028d3b4 SolveSpace`SolveSpace::AssertFailure(file="/Users/dgramop/Projects/solvespace/src/render/gl3shader.cpp", line=104, function="Co
mpileShader", condition="shader != 0", message="glCreateShader failed") at util.cpp:15:5
    frame #5: 0x000000010002aac4 SolveSpace`SolveSpace::CompileShader(res="shaders/edge.vert", type=35633) at gl3shader.cpp:104:5
    frame #6: 0x000000010002a798 SolveSpace`SolveSpace::Shader::Init(this=0x0000000128841674, vertexRes="shaders/edge.vert", fragmentRes="shaders/edge.frag", lo
cations=size=3) at gl3shader.cpp:135:19
    frame #7: 0x000000010002c96c SolveSpace`SolveSpace::EdgeRenderer::Init(this=0x0000000128841668, a=0x0000000128841610) at gl3shader.cpp:461:12
```

### Additional information
The version of solvespace I have "installed" still works (3.1-70bde63c). I first ran into the issue in the parameters branch, but then checked out master to see if it would happen again. I was able to compile on this mac a few months ago. Perhaps something in my packages changed

dgramop, 2024-03-24:
Just checked out the 70bde63c commit, and I'm unable to run this either, after compiling. The "installed" version seems to run just fine. Here's how I'm invoking cmake:
```
cmake -DCMAKE_BUILD_TYPE=Debug -DENABLE_OPENMP=ON ..
```

dgramop, 2024-03-24:
What's driving me crazy is that this did work earlier today, all until I re-ran cmake...

ruevs, 2024-03-24:
I've only tried to [build](https://github.com/solvespace/solvespace?tab=readme-ov-file#building-on-macos) the macOS version with a fairly old macOS and XCode.

- Some updates?
-  Do the [shaders](https://github.com/solvespace/solvespace/tree/master/res/shaders) get linked?
- Apple have been "talking" about deprecating OpenGL for a while... but the release works for you...
- Try if the artefact from the latest CI build on master works. https://github.com/solvespace/solvespace/actions/runs/7400855007

Perhaps @vespakoen can help you - he works on a Mac.

dgramop, 2024-03-24:
By some updates, I mean a minor version of MacOS and some fresh homebrew packages.

When I build the latest release myself I get the same problem. Doubt it’ll replicate with the CI prebuilt since I’m able to run the installed version, but I’ll check and report back.

I’m not actually sure how shaders are linked, how should I go about testing for this?

dgramop, 2024-03-25:
Just tested with the latest macOS artifact and it works. Definitely something wrong with how I'm building this

dgramop, 2024-04-02:
Wrote a dummy program and linked to the OpenGL framework:
```
clang -g -framework OpenGL test.c
```

Then ran `otool -L` to see what dynamic links it made:
``` 
./a.out:
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2
```

Looks like it's using openGL 1.0.0 from the system in the dummy program... which of course SIGSEGV's if I make any calls to functions in the gl3 headers, which happen to be provided in that same Frameworks folder (even though apparently version 3 is not available?)

Maybe somehow I lost openGL version 3... also interesting to see that came picks mesa libGL form home-brew when I had that installed. Maybe there's a difference between the version of libGL that gets headers compiled in and the one that gets dynamically loaded on my system....

dgramop, 2024-04-02:
[ 82%] Copying resource locales/zh_CN.po
[ 83%] Copying resource shaders/edge.frag
[ 83%] Copying resource shaders/edge.vert
[ 83%] Copying resource shaders/imesh.frag
[ 83%] Copying resource shaders/imesh.vert
[ 83%] Copying resource shaders/imesh_point.frag
[ 84%] Copying resource shaders/imesh_point.vert
[ 84%] Copying resource shaders/imesh_tex.frag
[ 84%] Copying resource shaders/imesh_tex.vert
[ 84%] Copying resource shaders/imesh_texa.frag
[ 84%] Copying resource shaders/mesh.frag
[ 86%] Copying resource shaders/mesh.vert
[ 86%] Copying resource shaders/mesh_fill.frag
[ 86%] Copying resource shaders/mesh_fill.vert
[ 86%] Copying resource shaders/outline.vert
[ 86%] Copying resource threejs/SolveSpaceControls.js
[ 86%] Copying resource threejs/hammer-2.0.8.js.gz
[ 87%] Copying resource threejs/three-r111.min.js.gz
[ 87%] Built target resources

^ in case this is what you mean by shaders being linked

Also maybe we should bring CI from macOS 12 to macOS 14

dgramop, 2024-04-04:
(Ladies and Gentlemen, We Got Him)[https://www.youtube.com/watch?v=RYonSOkZ5ZE]

This bug has been driving me nuts every time I feel like working on solvespace.

Digging around with `otool`, I took the known good & working builds from CI and did a diff between the dylibs for working vs good. 

This led me to uninstall libGL provided by Mesa from homebrew. After I did that, my build somehow tried linking to /opt/local/lib/libGL.1.dylib (which was not linked to by the known-good one). As a quick hack I edited `build/src/CMakeFiles/solvespace.dir/link.txt` (generated by cmake) and removed "/opt/local/lib/libGL.1.dylib". I ran make, opened it up, and my built Solvespace loaded up without crashing!

I owe this thread and future readers a more stable fix, so I'll do some further diagnosing, document the fix, and close the issue.

TL;DR: If this happens to you, consider comparing your build's dylib paths with the known good one. Make sure to not have Mesa installed (even though it may be used by solvespace on other platforms).

dgramop, 2024-04-04:
When running `cmake .. -DCMAKE_BUILD_TYPE=Debug -DENABLE_OPENMP=ON`
The culprit is:
```
....
-- Found OpenGL: /opt/local/lib/libGL.dylib
....
```
find_package(OpenGL REQUIRED) in CMakeLists.txt is what locates this file. That file seems to just exist on my mac (and as far I know shipped with my mac). It doesn't symlink anywhere to homebrew and I didn't manually build & install any openGL. I think my install of openCV is what created this mess. 

So TL;DR: All we need to do to link OpenGL is give Clang the  -framework OpenGL file, and nothing else. The additional stuff that find_package gives back can result in weird conflicting/broken links to libGL. Maybe there's something to be fixed within Cmake's FindOpenGL module itself. Till then we should just use -framework OpenGL

dgramop, 2024-04-07:
I got another apparent dynamic linker error when building on my mac (but not on prebuilt resources) for error dialogs. Will investigate this eventually

vespakoen, 2024-04-21:
Try compiling locally using the github scripts, it might help?

```sh
.github/scripts/install-macos.sh ci
.github/scripts/build-macos.sh release arm64
```

CI is up and running again for macOS so new pre-compiled binaries should be available soon...

PR #1502 (2024-12-10): Extlib fixups and macOS framework finding fix
1. Fix extlib (Python / JS bindings) PR remarks from #1343 1343 (mostly removing unused code)
2. Fix macOS framework finding (closes #1454 and #1456)
3. Publish npm package with "stable" label when build is tagged
closed 2024-12-10 by vespakoen