solvespace/solvespace #1458: Document Constraint Performance/Efficiency

samw3, 2024-04-23:
This is a documentation request.

I'm new to SolveSpace and still getting the hang of parametric modeling. I've noticed there are various ways to apply constraints, and the documentation highlights the Horizontal (H) and Vertical (V) constraints as preferable due to their ease of calculation. However, I couldn't find additional information on the performance and efficiency of different constraint types.

Could we possibly add some documentation explaining the relative performance and efficiency of the other constraints?  I feel it would help someone like me better learn the ropes of parametric modeling?

Thank you for the fantastic project!

ruevs, 2024-04-24:
I'm glad you like SolveSpace. The reference manual [correctly points out](https://solvespace.com/ref.pl#Horizontal) that horizontal and vertical constraints are indeed very "easy" to solve. They essentially boil down to `x1=x2` and `y1=y2` respectively.

This is the code that creates the equations that have to be solved:
https://github.com/solvespace/solvespace/blob/569ac27dd768612fb9b09b80295ef1439098defd/src/constrainteq.cpp#L858

The code for all other constraints are above and below it. The only other similarly simple constraint is `EQUAL_RADIUS`. All other constraints are more complex, which is not to say they are equally hard to solve.

I'm not sure whether it is worth making the documentation more complicated. It would be very hard to explain all the cases.

To demonstrate the above I just read through the code and sorted the constraints in groups from "fastest" to "slowest".
The list may not be correct since I did not spend enough time and have certainly not tested the performance.

But in general:

1. HORIZONTAL, VERTICAL, EQUAL_RADIUS
2. POINTS_COINCIDENT (in 3d), SYMMETRIC_HORIZ, SYMMETRIC_VERT, WHERE_DRAGGED
3. DIAMETER, PARALLEL
4. PT_PT_DISTANCE (could be very different based on the situation), PROJ_PT_DISTANCE, PT_PLANE_DISTANCE, PT_FACE_DISTANCE, EQ_LEN_PT_LINE_D, EQ_PT_LN_DISTANCES, LENGTH_RATIO, POINTS_COINCIDENT (in workplane), AT_MIDPOINT, SYMMETRIC_LINE, SAME_ORIENTATION, ARC_LINE_TANGENT, CUBIC_LINE_TANGENT (in 3d), CURVE_CURVE_TANGENT (curve)
5. PT_IN_PLANE, PT_ON_FACE, PT_ON_LINE, SYMMETRIC, CUBIC_LINE_TANGENT (in workplane), CURVE_CURVE_TANGENT (arc of circle)
6. PT_LINE_DISTANCE, EQUAL_LENGTH_LINES, LENGTH_DIFFERENCE, PT_ON_CIRCLE
7. ARC_ARC_LEN_RATIO, ARC_LINE_LEN_RATIO, ARC_ARC_DIFFERENCE, ARC_LINE_DIFFERENCE, EQUAL_LINE_ARC_LEN, PERPENDICULAR, ANGLE, EQUAL_ANGLE

As you see the list is... long... and includes many more things than visible in the user interface :-)

ruevs, 2024-04-24:
In fact I think that the constraints tutorial https://solvespace.com/constraints.pl provides a much more practical insight on how constraints work.

What I wrote above and this https://solvespace.com/tech.pl are "advanced topics" :-)

closed 2024-04-24 by ruevs

Evil-Spirit, 2024-05-21:
The difference of Horizontality, Verticality, Point-Coincident and Equal Radius constraints is fact that it having form of a = b, so inside solver this basically solved as variable substitution, so not taken part in the numerical Newton method algorithm, so it is very fast. And plus, I have optimized algorithm of substitution to work in linear time o(n)