solvespace/solvespace #1460: Font Kerning Not Working

kdickerson, 2024-05-25:
### System information

- **SolveSpace version:** 3.1~flatpak
- **Operating system:** Linux Mint 21.1

### Expected behavior

Sketching curve from text should apply kerning when rendering the text.

### Actual behavior

Kerning not applied, resulting in stilted spacing.

### Additional information

It looks like kerning works correctly on the built-in, default font, but I'm not sure it's working on external fonts (or at least doesn't appear to work on many).

I used FontForge to make some modifications to an existing font and then generated a new font.  I've tried all the options I can find when generating the font that pertain to kerning, but nothing I do seems to make any difference to SolveSpace.  Other applications successfully apply the kerning.

[image]

I've tried every combination of:
Format: TrueType, OpenType
Options:
- Old style 'kern'
- Windows-compatible 'kern'
- Prefer native kerning

Any advice on what settings I need in my font in order to get kerning to work correctly in SolveSpace?

phkahler, 2024-05-30:
My guess is that Solvespace doesn't have code to support kerning, but I could be wrong. The obvious work around is to place multiple text objects and position them manually.

kdickerson, 2024-05-30:
Ah, looks like you're correct.  Solvespace doesn't implement kerning.

But, what if it did? :-)

[image]

My C++ skills are decades out of date, but this (rather surprisingly) took me less than 3 hours to figure out and implement.  And is only 7 lines of code.  I'll submit a PR.

It can't support fancy, modern GPOS kerning without doing something much more drastic, but being able to support the older `kern` table kerning is a nice step forward and as much as I need since I can ensure the font I care about has that data when I generate it from FontForge.

And props for the build instructions and tooling actually being correct and painless to follow!

ruevs, 2024-05-30:
Indeed we use FreeType and it seems that ["FreeType only supports kerning via the (rather simple) ‘kern’ table."](http://freetype.org/freetype2/docs/glyphs/glyphs-4.html)

Until now I was not aware we did not use kerning - never looked at the "text" code, but it is possible that this could have been intentional - kerning can make glyphs overlap and then the resulting sketch is self intersecting and will not form a solid when extruded. See here https://github.com/solvespace/solvespace/issues/1152 and here https://github.com/solvespace/solvespace/issues/285

So the splitting and merging of glyphs should be implemented and/or kerning should be an option - perhaps on each individual "text" entity.

kdickerson, 2024-05-30:
Not sure where you prefer discussion to continue at this point, but I'll do so on the PR.

kdickerson, 2024-06-01:
To help anyone stumbling into this in the future: With the proposed change, fonts containing an old-style `kern` table can apply kerning as an option in SolveSpace.  In FontForge this means generating a font with the option "OpenType -> Old style 'kern'" selected. This works whether the actual file format is TTF or OTF.

ruevs, 2024-06-11:
Merged here 6b79e69d60b24d5e9c2dfe55b58edb1ebd4e0aa7

closed 2024-06-11 by ruevs