solvespace/solvespace #1465: "it can only be saved on the same volume."

wpwrak, 2024-06-12:
### System information

3.1-40a1fdc6
Ubuntu 22.04.4

### Expected behavior

I have two SolveSpace models, a case top (top.slvs) and the case bottom (bottom.slvs), where the latter links the former, for reference. Both files are in the same directory. I try to save the latter, expecting this to just work, as it did when I worked on similar models in the past.

### Actual behavior

I get an error message "This sketch links the sketch 'top.slvs'; it can only be saved on the same volume." and nothing is saved.
Looking at the code, I think "volume" refers to a DOS-type "drive", i.e., an independent root of a directory hierarchy.

### Additional information

In bottom.slvs, the case top is referenced with
Group.impFileRel=top.slvs
In Path::RelativeTo  both path, top.slvs and bottom.slvs, consist just of the file name, without any directory components.

The problem seems to be caused by Path::RelativeTo calling Path:Expand with the default (false) for fromCurrentDirectory, which prevents adding the current directory path.

I could make it work by simply adding
`
if (!strchr(raw.c_str(), '/'))
    fromCurrentDirectory = true;
`
at the beginning of Path::Expand, but I'm not sure if that's the right way to solve this.

ruevs, 2024-06-17:
This works just fine on Windows. Someone on Linux needs to debug and test this.

ruevs, 2024-06-17:
I "recently" ( :-D ) touched the file path handling logic here:
https://github.com/solvespace/solvespace/commit/d5e8a8267cefe84f15541249ab886aead94c0935
and here:
https://github.com/solvespace/solvespace/commit/6f7e45ba9f6d25079d48f1f14564b1a0b8803c85

I wonder if that caused the problem?

WickedSmoke, 2024-09-16:
This problem occurs when the main model is loaded from the CLI with no path (or `./`).

The intended behavior of `Path::RelativeTo` is not clear when used on two files in the same directory.  Is this function supposed to be returning "" or "./" on UNIX?

Is the RelativeTo check in `SolveSpaceUI::SaveToFile` really only needed on Windows (as there are no "volumes" on UNIX)?

ruevs, 2024-09-19:
Fixed here https://github.com/solvespace/solvespace/pull/1485

WickedSmoke, 2024-09-20:
The error still appears with commit a31a89a2.

I'll repeat the two issues I see here:
  1. A "same volume" check should not be done on UNIX.
  2. The relative path handling is not properly handling empty paths.
 
When the "same volume" check is commented out saving fails with a  "Cannot relativize path" assert (with or without a31a89a2).

ruevs, 2024-09-20:
@WickedSmoke OK I guess GTK does not expand the path on dialog->RunModal() (Windows does). Try this one please https://github.com/solvespace/solvespace/pull/1486

WickedSmoke, 2024-09-20:
This has nothing to do with file dialogs (I'm running the Qt port) as the code should handle paths in any form.  Here's how to reproduce this:

1. Get the files from issue #1474.
2. From the CLI run `solvespace feet-10deg.slvs`.
3. Do a "File -> Save".

I expect the same results will occur on all platforms.  I'll check your latest patch.

Even on Windows I'm not sure what the "same volume" check is supposed to be doing.  There's the path of the main model (absolute or not), and any other relative path should be using that, right?  How would you ever get relative paths with a different drive letter?

ruevs, 2024-09-20:
> This has nothing to do with file dialogs (I'm running the Qt port) as the code should handle paths in any form. Here's how to reproduce this:
> 
>     1. Get the files from issue [Can't open files containing linked objects in version 3.1 #1474](https://github.com/solvespace/solvespace/issues/1474).
> 
>     2. From the CLI run `solvespace feet-10deg.slvs`.
> 
>     3. Do a "File -> Save".

This is what I did. And it was broken on Windows - so I fixed it in https://github.com/solvespace/solvespace/commit/a31a89a2f5620d3fac4ea4aa83119a8ae08f014a. Now it works.

> I expect the same results will occur on all platforms. I'll check your latest patch.

I guess not - because it still fails for you. The only explanation is that this code:

https://github.com/solvespace/solvespace/blob/a31a89a2f5620d3fac4ea4aa83119a8ae08f014a/src/solvespace.cpp#L587

does not expand `newSaveFile`, which starts as "feet-10deg.slvs", to a full path on QT (and perhaps not on GTK). Windows does expand it and the path comes back "c:\blahblah\blah\whatever\feet-10deg.slvs", which then allows the  ["same volume check"](https://github.com/solvespace/solvespace/blob/a31a89a2f5620d3fac4ea4aa83119a8ae08f014a/src/file.cpp#L288) to work. As a test I intentionally hacked up the code to leave only the file name before the same volume check and it breaks. That is why I made https://github.com/solvespace/solvespace/pull/1486 - please test it.

> Even on Windows I'm not sure what the "same volume" check is supposed to be doing. There's the path of the main model (absolute or not), and any other relative path should be using that, right? How would you ever get relative paths with a different drive letter?

Exactly. There can not be a relative path to a file on a different drive (c:, d:) on Windows, and since the linked files are stored as relative paths inside the .slvs file they can not be on a different drive on Windows. This is why the check is there. And it should always succeed on Unix like systems. For reference https://github.com/solvespace/solvespace/issues/1423

WickedSmoke, 2024-09-20:
The CLI test is using "Save", not "Save As", so the dialog->RunModal() path is not executed.  The e5afcdded patch causes the "Cannot relativize path" assert.

I really think the focus needs to be on the behavior of Path::RelativeTo.  I would expect an empty path to be valid (indicating the same directory), but this is being treated as an error condition.

ruevs, 2024-09-20:
> The CLI test is using "Save", not "Save As", so the dialog->RunModal() path is not executed. The [e5afcdd](https://github.com/solvespace/solvespace/commit/e5afcdded767689c3588079ec20866ecc64a169a) patch causes the "Cannot relativize path" assert.

Offf, of course, please try again with 21621dbf3ba70068f840ddd966ecd4aaec622e5a - I force-pushe the branch.

> I really think the focus needs to be on the behavior of Path::RelativeTo. I would expect an empty path to be valid (indicating the same directory), but this is being treated as an error condition.

RelativeTo is supposed to return the path of `this` relative to the `const Path &base`. An empty string really means "it is not possible to relativise the path` and this should only rally happen on Windows when the paths are on different drives. Otherwise you are right - the "proper" thing to fix is the RelativeTo function by `Expand`ing inside it "properly" - but see my [comment above](https://github.com/solvespace/solvespace/issues/1465#issuecomment-2363851169).

WickedSmoke, 2024-09-23:
The SaveUsingTable patch in 21621dbf allows the feet-10deg.slvs file to be saved on Linux.

ruevs, 2024-09-24:
Thank you @WickedSmoke .

ruevs, 2024-09-24:
https://github.com/solvespace/solvespace/pull/1486 merged in 5528da7ffbfe511a90c46864da40eec5305a89ae

commit a31a89a2 (2024-09-19): Fix saving assemblies when opened with a relative path on the command line.

When a file containing linked files (an assembly) was opened by passing
a relative path to it on the command line; saving it would fail with an
error:
"This sketch links the sketch '%s'; it can only be saved on the same volume."

The problem stems from not `Expand`-ing paths immediately in `main` which
was introduced in 3ea8ebf see here for details:
https://github.com/solvespace/solvespace/issues/1474#issuecomment-2361853738

The fix is to expand paths to linked files before making them relative to
the assembly.

Fixe [... cut]
PR #1485 (2024-09-19): Fix saving assemblies when opened with a relative path on the command…
… line.

When a file containing linked files (an assembly) was opened by passing a relative path to it on the command line; saving it would fail with an error:
"This sketch links the sketch '%s'; it can only be saved on the same volume."

The problem stems from not `Expand`-ing paths immediately in `main` which was introduced in 3ea8ebf see here for details:
https://github.com/solvespace/solvespace/issues/1474#issuecomment-2361853738

The fix is to expand paths to linked files before making them relative to the assembly.

F [... cut]
closed 2024-09-24 by ruevs