solvespace/solvespace #1466: constrain symmetric causing "constraints are incompatible"

wolfv6, 2024-06-15:
### System information

- **SolveSpace version:** 3.1~70bde63c

- **Operating system:** Fedora 40

### Expected behavior

The assembly should have constrained without error. 

### Actual behavior

The error message in the Property Browser said:
the following constraints are incompatible
- same-orientation
- pt-on-line

### Additional information

The error is reproduced in the attached [box_on_foor.zip](https://github.com/user-attachments/files/15849354/box_on_foor.zip) > box_on_floor-asm_error.slvs
I think the error has something to do with the "constrain symmetric" in box.slvs

Constraining the other and of the box in the same way works fine in box_on_floor-asm.slvs

On https://solvespace.com/forum.pl?action=viewthread&parent=5754, Tom said it worked for him using 3.0, so maybe it's a new bug?

wolfv6, 2024-06-15:
Also getting same error constraining assembly that does not use "Constrain symmetric".

The error is reproduced in the attached [box_on_foor_3.zip](https://github.com/user-attachments/files/15854970/box_on_foor_3.zip) > box_on_floor-asm_error.slvs

Constraining a different point in the same way works fine in box_on_floor-asm.slvs

ruevs, 2024-06-17:
This is a duplicate to #246 #1105 #1354

It boils down to the solver having "convergence" problems with "large" objects. If you decrease the size of box.slvs to 127x40 and the size of floor.slvs to 152 it works fine.

There was a discussion on this topic recently on the forum https://solvespace.com/forum.pl?action=viewthread&parent=5740&tt=1718184803

wolfv6, 2024-06-17:
Thanks for the links.

The dimensions of the box and floor where already smaller than 127x40 and 152 x 152 (inches).
What units are you using?

ruevs, 2024-06-17:
Millimeters - internally everything is in millimeters :-)

wolfv6, 2024-06-19:
A simple work around is to use mm or .1-inch were inches are meant.
Should I close this issue?

ruevs, 2024-06-20:
Let it stay open. We can easily close them all when this is fixed.