solvespace/solvespace #1467: "Regenerate All" not working on nested assemblies

wolfv6, 2024-06-19:
### System information

- **SolveSpace version:**  3.1~70bde63c
- **Operating system:** Fedora 40

### Expected behavior

I have a project containing a hierarchy of assemblies, each assembly being dependent on a previous assembly.
I edit and save a dependent part .slvs file.
Then I open the assembly .slvs file and click "Edit → Regenerate All".
I expect the assembly to display with the new version of the part .slvs file.

### Actual behavior'

"Edit → Regenerate All" didn't seem to do anything.

### Additional information

Simplified example in [nested_assemblies.zip](https://github.com/user-attachments/files/15906071/nested_assemblies.zip)

Similar posts:
https://solvespace.com/forum.pl?action=viewthread&parent=5769&tt=1718822563
https://solvespace.com/forum.pl?action=viewthread&parent=2777&tt=1563959569

"Regenerate All" is mentioned in:
https://solvespace.com/ref.pl
https://solvespace.com/box.pl

BKLronin, 2024-06-19:
I just hit space usually.

wolfv6, 2024-06-20:
So "Regenerate All" works on your system?  What version of solvespace are you using?

Space is the shortcut for "Regenerate All", which does nothing on my system.
I test Regenerate All" on this simple example: [nested_assemblies.zip](https://github.com/user-attachments/files/15906889/nested_assemblies.zip)

BKLronin, 2024-06-20:
Yes, Iḿ on the latest Linux stable release 3.1 ish.

With the three wndows open:
- Change lowest part in hirarchy "traingle-part.slvs 
- Save
- Go up in hirarchy press space ( if window already open)
- Save
- In top level press space -> Changes updated.

pixelEDI, 2024-06-20:
If you make a change in "triangle-part" - save - and switch to "rectangle-triangle+triangle-asm," it won't update. But if you instead update "triangle+triangle-asm" before updating "rectangle-triangle+triangle-asm," then the update works.

As far as I understand, this is the issue with the nested hierarchy, right?

I'm also using the 3.1 Linux release.

BKLronin, 2024-06-20:
Yes, 
does make sense because the triangle+triangle-asm.slvs still has the old state.

wolfv6, 2024-06-20:
@pixelEDI
Yes, I was expecting the "Regenerate All" function to traverse the dependency hierarchy depth-first, regenerating and saving each file on the way up.
The "All" in "Regenerate All" is misleading.
Consider removing the "All".

BKLronin, 2024-06-20:
The "all" refers to all imported parts on the same hirarchy level I suppose.

ruevs, 2024-06-20:
#287

closed 2024-06-21 by ruevs