solvespace/solvespace #1472: How to make threads correctly?

lukexi, 2024-07-18:
### System information

- **SolveSpace version:** Version 3.1 (3.1~70bde63c)
- **Operating system:** macOS 14.5

### Expected behavior
I'm trying to make simple screws and threaded holes (by subtracting screws), and I haven't found the formula yet that doesn't result in a pile of interfering parts / naked edges.

The method that seems most intuitive is to make a box-plus-triangle shape which captures the major and minor diameters of the screw, and then make a helix from it with a given pitch. I'd like to assemble this into other models as a difference group to get threaded holes.

### Actual behavior
I end up with many interfering parts/naked edges unless I make sure the helix doesn't self-intersect (which wouldn't result in a solid screw shape). I've also tried helixing just the triangle portion for the threads and then forming the core with a unioned extruded circle but this gives similarly broken results. "Force NURBS to triangle mesh" does not resolve the errors.

### Additional information
I hoped to find the answer looking at examples but I haven't found one yet — e.g. #793 includes a collection of M3, M4 etc. bolts which work very well, but when I try to look at their construction I only see a blank in the "thread cross section" group. (I've tried opening them with 3.0 as well — 2.3 is too old to open them).

This at least proves it's possible to do : ) — can anyone explain what I'm missing? Thank you!!

Here's a very simple example as described in the first section.

[Helix_Example.slvs.zip](https://github.com/user-attachments/files/16284688/Helix_Example.slvs.zip)

phkahler, 2024-07-19:
I just opened that old parts_bin zip file. Using a recent version of Solvespace many of the bolts work but some do not. M3 and M5 a lot of problems with naked edges. But several of them did open fine - I have my chord tolerance set to 0.1 percent on the configuration screen.

That approach to making threads is not great. Making the cross section is hard - you need a curve that changes radius linearly with angle and I tried to approximate that and it looks very strange. It does make that bolt shaft with a single helix group though when it works.

People have had some success making a cylinder and either adding or cutting threads staring with a triangle. The trick there is that the helix can not self-intersect, and the inner flat surface of the triangle can not be coincident with the surface of the cylinder. You have to overlap the two groups a little and make a union. Chord tolerance is also likely to affect it, but less so than the other construction approach.

lukexi, 2024-07-20:
(I was able to open the parts_bin and see the final 3D model for e.g. M3 just fine, I just wasn't able to set the "active" dot to "g002-thread_cross_section" to see the initial sketch to learn its approach)

Aha, starting with a cylinder and cutting threads with a non-intersecting helix sounds like a great way to go, thanks so much Paul!

I wonder if there's a good unified place to document things like this, perhaps enabling the Wiki feature on this repository?

lukexi, 2024-07-20:
I was just able to test this out and confirm that starting with a cylinder and then carving threads with a triangle helix works beautifully, thanks again!!

ruevs, 2024-07-20:
> I wonder if there's a good unified place to document things like this,

We label issues with useful discussions on using SolveSpace with the "documentation" label. https://github.com/solvespace/solvespace/issues?q=is%3Aissue+label%3Adocumentation

Evil-Spirit, 2025-03-11:
Have you moved to OpenCASCADE already?

nobodyinperson, 2025-03-11:
No, why? Look at Dune3D, which uses OpenCASCADE and the SolveSpace solver.

phkahler, 2025-03-11:
I kind of wish Dune3D would have first put their GTK4 interface on Solvespace and then replaced our NURBS with OpenCASCADE. That would have been better for us, but may have been more work for them. We could have opted out of the OCCT change or maybe made it an option. The way to do it is to replace SShell objects with the OCCT equivalent. You'd keep Solvespace curves and entities for sketching. It would require converting our curves to their curves when it's time to make extrusions. OCCT could also be used to create triangle meshes for us and hand them back.

ruevs, 2025-03-12:
> I kind of wish Dune3D would have first put their GTK4 interface on Solvespace and then replaced our NURBS with OpenCASCADE. That would have been better for us, but may have been more work for them.

The way it actually happened is better. In my opinion having only one open source NURBS [CAD kernel](https://en.wikipedia.org/wiki/Computer-aided_design#CAD_kernels) is not nice (just like [browser engines](https://en.wikipedia.org/wiki/Browser_engine#Notable_engines)). Ours is limited compared to OpenCASCADE but it is arguably cleaner and easier to understand. It deserves to exist and hopefully get improved in time.