solvespace/solvespace #1474: Can't open files containing linked objects

davidgiven, 2024-07-21:
- **SolveSpace version:** 3.1+ds1-3.1+b1
- **Operating system:** Linux Debian testing

If I try to open a file containing a linked object, the linked object doesn't load, and all the constraints attached to the linked object break:

[image]

Examining the loaded file, it seems that the filename for the linked object hasn't been read properly out of the slvs file:

[image]

If I run solvespace with strace, this is corroborated. I can see it load the original file, then it tries to open `.`:

```
...
read(16, "2031085529958 1.9999999999999997"..., 4096) = 4096
read(16, "t 0 -24.24264068711928388211 -4."..., 4096) = 4096
read(16, "8024742503792481330\nAddCurve\nCur"..., 4096) = 4096
read(16, "7.71427983280334217397 Weight 1."..., 4096) = 4096
read(16, "0637458226738 47.720394026388149"..., 4096) = 533
read(16, "", 4096)                      = 0
close(16)                               = 0
openat(AT_FDCWD, ".", O_RDONLY)         = 16
newfstatat(16, "", {st_mode=S_IFDIR|0775, st_size=4096, ...}, AT_EMPTY_PATH) = 0
read(16, 0x560f0e262d60, 4096)          = -1 EISDIR (Is a directory)
close(16)                               = 0
...
```

I noticed this with new files, i.e. the ones I'm currently working on, but trying to open some old ones which used to work also fails, so I think this is a new regression?

I'm enclosing the two files.

[files.zip](https://github.com/user-attachments/files/16323800/files.zip)

davidgiven, 2024-07-21:
Additional data: opening the file with "Open recent..." or on the command line causes the problem to manifest. However, if I open the file with "Open..." and selecting the filename manually, it seems to load fine! It seems quite repeatable.

ruevs, 2024-07-24:
Works on Windows. Please post the result from going to `Help | Go to GitHub commit`

davidgiven, 2024-07-24:
It's https://github.com/solvespace/solvespace/commit/70bde63c, which I now realise is really old, but that's the current version in Debian.

Looking at the Debian changelog, a new version came out in 2024:

```
solvespace (3.1+ds1-3.1) unstable; urgency=medium

  * Non-maintainer upload.
  * Rename libraries for 64-bit time_t transition.  Closes: #1062911

 -- Benjamin Drung <bdrung@debian.org>  Thu, 29 Feb 2024 17:02:12 +0000
```

This seems the most plausible candidate for whatever's gone wrong, given that it worked for me before, but I don't know why this would have broken something.

ruevs, 2024-07-24:
70bde63c is the last official release 3.1
https://github.com/solvespace/solvespace/commits/v3.1/

But this rules out my changes from last November that I mentioned in #1465 as the potential problem...

It is also VERY surprising that no one has caught this on Linux for two years?! @phkahler you are on Linux - can you reproduce this problem?

WickedSmoke, 2024-09-16:
The `feet-10deg.slvs` file loads fine with 9dc81f7d + Qt changes on Fedora via "Open Recent" and CLI argument.

WickedSmoke, 2024-09-18:
The problem also occurs on Windows in v3.1.  Did you run `solvespace feet-10deg.slvs` @ruevs?

davidgiven, 2024-09-18:
Oh, thank goodness --- it's not just me...

ruevs, 2024-09-18:
I'll try from the command line and debug it tomorrow.

ruevs, 2024-09-19:
When running (from the commend line) `solvespace.exe feet-10deg.slvs` with the two files (`feet-10deg.slvs` and `m4nutcylinder.slvs`) in the current directory [version 3.1](https://github.com/solvespace/solvespace/releases/tag/v3.1) fails to load the linked file. Current master loads it just fine on Windows because this was fixed. But it is still broken on Linux.

The history is long, ugly and complicated:

First [while fixing](https://github.com/solvespace/solvespace/pull/740) this https://github.com/solvespace/solvespace/issues/622#issuecomment-710020308 I broke it for non-Windows systems in this commit: https://github.com/solvespace/solvespace/commit/3ea8ebfaf57ab061f18f2838f6599d10a339d59b - by removing `Expand` from [`main`](https://github.com/solvespace/solvespace/blob/9fb6ade06bc271ba1196b79a6ed3d1eebcea9609/src/platform/entrygui.cpp#L24) but adding it in [`OpenFile`](https://github.com/solvespace/solvespace/blob/9fb6ade06bc271ba1196b79a6ed3d1eebcea9609/src/platform/platform.cpp#L414) **_only_** for Win32. But no one noticed for a long time...

Later on I actually "fixed" the opening of assembly files with relative paths here:
https://github.com/solvespace/solvespace/commit/d5e8a8267cefe84f15541249ab886aead94c0935
related discussion: 
https://github.com/solvespace/solvespace/pull/1421
https://github.com/solvespace/solvespace/pull/1194 
https://github.com/solvespace/solvespace/issues/1292

And finally another bug with linked files fixed here:
https://github.com/solvespace/solvespace/commit/6f7e45ba9f6d25079d48f1f14564b1a0b8803c85
related discussion:
https://github.com/solvespace/solvespace/issues/1347
https://github.com/solvespace/solvespace/issues/1347#issuecomment-1448979455

So finally the question remains why did I leave Linux (and anything not Windows) "behind"/broken here?
https://github.com/solvespace/solvespace/blob/9fb6ade06bc271ba1196b79a6ed3d1eebcea9609/src/platform/platform.cpp#L414
I don't remember - probably because this cross platform path handling is rather "baroque" and I was afraid of breaking something?

The whole history
https://github.com/solvespace/solvespace/commits/master/src/platform/platform.cpp

The fix is trivial - I'll make a pull request - but you guys will have to test it on Linux. My Raspberry Pi is busy with other stuff :-)

rpavlik, 2024-09-20:
I totally do not remember this, I use assemblies like all the time. Happy to test on Linux.

phkahler, 2024-11-26:
Is this fixed?  @rpavlik have you had a chance to test it?

ruevs, 2025-05-17:
@iscgar you are on Linux and currently actively working on SolveSpace. Would you please test/confirm that this is fixed, so that I can close it?

iscgar, 2025-05-17:
Working fine with latest master (60d0dab0b62cd50f01981a4669233aca9f4f1806), both when running `cd <solvespace-build>/bin && ./solvespace ~/files/feet-10deg.slvs` as well as when running `cd ~/files && <solvespace-build>/bin/solvespace feet-10deg.slvs` (reading your description, it wasn't clear to me whether the problem is related to cwd handling or not, so I tested both cases just to be sure).

ruevs, 2025-05-18:
Thank you. What you did is enough. The relative path to the slvs file on the command line used to caue the problem, not the CWD.

closed 2025-05-18 by ruevs