solvespace/solvespace #1476: Improving boolean operations with ManifoldCAD

dgramop, 2024-08-04:
Several cad programs (like OpenSCAD) have started using ManifoldCAD as a backend for boolean operations, since it guarantees that if the inputs are manifolds, the output will be a manifold. 

There are some instances where solvespace will boolean two manifolds, and output a non-manifold surface (forcing me to check the force triangles to surface mesh option). I was wondering what we thought of switching to Manifold as a boolean engine in solvespace

### System information
MacOS M1. HEAD on Parameters branch

### Expected behavior
Basic boolean operations of manifold surfaces should produce manifold outputs

### Actual behavior
Non-manifold outputs, and solvespace complains during a render 

### Additional information
This is a general pattern

phkahler, 2024-08-07:
This a duplicate of #1371 where the author of Manifold also commented.

I'm no longer fan of this idea but I'm not opposed to it either. First, it's only for triangle meshes and we need to fix our NURBS boolean bugs either way. It would still be good to have better triangle handling but less important once we fix NURBS. Manifold also has a number (not too large) of dependencies which means our build process gets more complicated, and we have somewhat limited support in that area as it is (who will stick around to maintain it). Then there is the license, which is "compatible" but I'll leave it to others to figure out what that entails in terms of obligations. The dependencies are also not all under the same license, so that needs to be figured out.

closed 2024-08-07 by ruevs

dgramop, 2024-08-08:
Ah my bad, I suspected somebody would've brought this up before, and searched for it.. but I only searched open issues. 
In any case, clearly there's already been plenty of discussion here... doubt I have anything new to add.