solvespace/solvespace #1477: Developer onboarding

jonathan-strange, 2024-08-17:
I'm looking to get involved in the project however the code base is daunting, especially coming back to C++ after 20 years. I asked on the IRC channel for any "onboarding docs" and I was directed here. I'm looking for the type of information you'd get from a colleague the first week in the job:

- how the code is organised
- overall design goals
- particular areas of interest
- where to start :-)
- ...

I'm not a big fan of documentation for the sake of it, but it would be nice when starting out on the project to approach it less like a black box :-).

I'll be hanging around the IRC channel (raevn) in case someone on the team would be able to help out.

ruevs, 2024-08-17:
https://github.com/solvespace/solvespace/issues/624, https://github.com/solvespace/solvespace/issues/1222 , https://github.com/solvespace/solvespace/issues/1303

I would add https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt

It would also be useful to browse through the [issues labeled `documentation`](https://github.com/solvespace/solvespace/issues?q=label%3Adocumentation) most of them are related to using SolveSpace but some are about the architecture or implementation details.

closed 2024-08-19 by ruevs

jonathan-strange, 2024-08-19:
Thanks for the info @ruevs. I'll see if I can get my head around it.

phkahler, 2024-08-20:
You may also want to check things tagged [good first issue](https://github.com/solvespace/solvespace/issues?q=is%3Aissue+is%3Aopen+label%3A%22good+first+issue%22).  Not much is going to be easy in Solvespace, but I tag things that might be a smaller scope for someone new so they can start poking around in a specific area. Always feel free to ask questions on the issue tracker. Sometimes it's surprising the stuff people are actually reading.