solvespace/solvespace #1488: Mouse scroll direction is reversed in Solvespace

mkyral, 2024-10-06:
### System information

- **SolveSpace version:** 3.1~5528da7f (compiled from source)
- **Operating system:** Gentoo, gtkmm-3.24.9

### Expected behavior

Mouse scroll direction should be consistent with other apps.

### Actual behavior

I'm using latest Solvespace compiled from git. I did not used it for several weeks. Yesterday I've discovered, that scrolling direction in Solvespace is reversed and it is really annoying.

Initially I trough that there was some change in GTK, but inkscape, that also uses gtkmm, works normally. Also Solvespace 3.1 with the same gtkmm works correctly. So change is probably in Solvespace code.

### Additional information

-- Checking for modules 'gtkmm-3.0>=3.18;pangomm-1.4;x11'
--   Found gtkmm-3.0, version 3.24.9
--   Found pangomm-1.4, version 2.46.3
--   Found x11, version 1.8.10

ruevs, 2024-10-06:
Indeed. Someone reported the same on the forum https://solvespace.com/forum.pl?action=viewthread&parent=5835&tt=1728124046

Since you are compiling from source please change this line:

https://github.com/solvespace/solvespace/blob/5528da7ffbfe511a90c46864da40eec5305a89ae/src/platform/guigtk.cpp#L581

to `delta = -dy;` and see if it fixes the problem.

ruevs, 2024-10-06:
For cross reference https://github.com/solvespace/solvespace/pull/1464 https://github.com/solvespace/solvespace/pull/1470 https://github.com/solvespace/solvespace/commit/31d0c27ecdc4ace0a1793191629d258cbd861826 https://github.com/solvespace/solvespace/commit/a17fb155ab807bc9d2c26a024bfc8083679ad2eb

mkyral, 2024-10-06:
> Indeed. Someone reported the same on the forum https://solvespace.com/forum.pl?action=viewthread&parent=5835&tt=1728124046
>
> Since you are compiling from source please change this line: [...]

Yes, this fix the issue.

ruevs, 2024-10-07:
Fixed here c619fa9f1f5eaaa60eea0f08770a04e703e51115.

commit c619fa9f (2024-10-07): GTK: Fix wrong direction of zoom when "smooth scrolling" with mouse wheel

... on Linux/GTK

The bug was introduced here 31d0c27ecdc4ace0a1793191629d258cbd861826
https://github.com/solvespace/solvespace/pull/1464
https://github.com/solvespace/solvespace/pull/1470

Fixes #1488
closed 2024-10-07 by ruevs