solvespace/solvespace #1490: Can't add diameter or radius using Parameters branch

phkahler, 2024-10-25:
### System information

- **SolveSpace version:**  The parameters branch
- **Operating system:**  e.g. Fedora 40

### Expected behavior

When I select a circle and press "d" and type a number it should be used as the diameter

### Actual behavior

Technically this is on my fork with a little added to the branch, but nothing in the area that would cause this.

### Additional information

This is on the parameters branch.

phkahler, 2024-10-25:
@dgramop I think this is a regression.

dgramop, 2024-10-26:
Yeah I noticed this too, will figure out the conditions to replicate. It doesn't seem to always happen. I bet it's something stupid.

The (embarrassing) workaround I did in those cases was to just use a construction line 😬