solvespace/solvespace #1492: Circle tangent to line as construction tool

hofmannu, 2024-11-05:
### System information

- **SolveSpace version:** 3.1~70bde63c
- **Operating system:** ArchLinux

[image]

### Expected behavior

If I add the constraint that the circle should be tangent to the two lines, the sketch is fully defined and the through hole will have ensured equidistance from the two object edges.

### Actual behavior

I get an "Bad selection for tangent/parallel constraint" error.

### Additional information

Attached is the tiny sketch.
[circle_tangent_to_line.zip](https://github.com/user-attachments/files/17627847/circle_tangent_to_line.zip)

phkahler, 2024-11-05:
@hofmannu  You've found the "correct" way to make a line and circle tangent with what we have.  If someone does implement line/circle tangency I think it would be good to automatically add the point where they meet.

Since you're a new user and the diagram looks a little odd I'm going to guess that maybe your green circle is simply a way to specify a distance (radius) from the center of the white circle. If that's the case, you can just pick the point and one of the lines and press D to set the point-line distance. Do that again for the other point-line distance. The number of constraints available in Solvespace is quite large ;-)  You can even select pt-line and another pt-line and use eQual to set two pt-line distances to be equal while leaving the actual distance unconstrained. Unfortunately that doesn't work when the points are the same (circle center in your case).

hofmannu, 2024-11-05:
thanks a lot for your answer!

while i am new to this software, i have a background in mechanical engineering and used CAD for more than a decade by now. that tangent circle is quite a common step in the construction process. the two distance approach won't have the two lengths  mapped to each other? so i need to add the number twice or again make two lines, enforce equal length and perpendicularity which is a bit cumbersome even with the nice workflow of the interface and the keyboard shortcuts.

Maybe i will try to implement the constraint type myself as a first contribution ;)

phkahler, 2024-11-05:
> Maybe i will try to implement the constraint myself as a first contribution

I don't think any of our constraints add sketch entities, so this is unusual - it would be adding both. I'm not even sure what the best way to constrain the point would be - on circle and on line, or on circle and forming a perpendicular to the line and circle center. Then there is the visual representation of this constraint. It's a good project if you want to get involved!

BTW sometime a constraint internally generates different equations depending what kind of entities are being constrained. For example the Distance constraint is quite different for pt-pt, pt-line, pt-plane, pt-face. So mathematically this could be a perpendicular constraint *equation* and a pt-on-circle *equation*.

I had another idea that would also create a point - I'd like to select a line segment and press "m" for midpoint to have a new point added and a constraint to place it on the line midpoint.  @ruevs says this goes against the simplicity and consistency of Solvespace, and he's right about that.

ruevs, 2024-11-06:
> Btw: I discovered solvespace yesterday and am working with other feature overblown and unusable CAD software for Linux since three years. You guys are amazing and the tool you made here is honestly one of my favorite pieces of software I discovered in the past 5 years for Linux, well done <3

I'm glad you like it :-)

Reminded by your comment in #937 I started implementing it. It will not cover line segments tangent to circles (only arcs and splines) so if you want to get involved it is all yours :-)

hofmannu, 2024-11-06:
I did a first sketch for that very particular case of a circle and a line and a neat approach is if you solve the math for the line and the circle having excatly one intersection point. because that is the tangent case in the end. A line can intersect a circle at 0, 1, or 2 places and only with 1 you have a tangent. 

If you solve the equations for the intersection points of a circle and a line, you will get to a quadratic formula with the famous $\pm \sqrt{b^2 - 4ac}$ part. hence if $0 = b^2 - 4ac$, we will get the tangent case. I might need to double check if I did the math correct but I think that boils down to a simple relationship between direction of line $\vec{d}$ (assuming unit length), point on line $\vec{o}$, center position of circle $\vec{c}$ and radius of circle $r$ that must be fulfilled to have a tangent line on a circle / arc

$$
d_x \cdot (o_y - c_y) - d_y \cdot (o_x - c_x) = r
$$

but i will need to check now first how constraints are implemented in solvespace :D

ruevs, 2024-11-06:
@hofmannu #937 is implemented in https://github.com/solvespace/solvespace/pull/1493
you can try it if you build from source. Since you are on Linux it is very easy.

ruevs, 2024-11-07:
> but i will need to check now first how constraints are implemented in solvespace :D

@hofmannu Hopefully you will be pleasantly surprised :-)

First you need toi ad da new constraint type. For example `CIRCLE_LINE_TANGENT`, search for `ARC_LINE_TANGENT` to find the 10 places where you may need to add something (in reality 7).

You will also need to write two equations:

1. The line is perpendicular to the radius
2. The distance from the center of the circle to the line is equal to the radius

For 1. take a look at this (where `Dot` is the dot product):

https://github.com/solvespace/solvespace/blob/bb160669721e10096dc17427422ea465a7ec2b72/src/constrainteq.cpp#L936

For 2. look here:

https://github.com/solvespace/solvespace/blob/bb160669721e10096dc17427422ea465a7ec2b72/src/constrainteq.cpp#L293

and here:

https://github.com/solvespace/solvespace/blob/bb160669721e10096dc17427422ea465a7ec2b72/src/constrainteq.cpp#L570

for ideas. You may (or may not) have to use `CircleGetRadiusNum`  ;-)

And that's it™ ;-)

hofmannu, 2024-11-07:
I did the first draft will send you a pull request tonight :)

phkahler, 2024-11-07:
I just realized... The constraint equation for this could be to set the point-line distance equal to the circle radius. So check the point-line constraint equation and put the circle radius in place of the dimension (distance to line). Not sure this would be as robust as something more explicit about the point of tangency.

hofmannu, 2024-11-10:
I only need to position the labels correctly, then i am ready :)

https://github.com/user-attachments/assets/daa29972-ea18-465d-a775-24e69e57b266

dadoprom, 2024-12-18:
Please finish this :)

Evil-Spirit, 2025-03-11:
You guys don't need to do any work, just copy-paste all the things from NoteCAD where all cornercases was solved already.

nobodyinperson, 2025-03-11:
Go for it if it's that easy.

ruevs, 2025-03-11:
It is easy to implement - see [my comment above](https://github.com/solvespace/solvespace/issues/1492#issuecomment-2461732091) but I am not at all convinced, that we need this new constraint.

> You guys don't need to do any work, just copy-paste all the things from NoteCAD where all cornercases was solved already.

Presumably @Evil-Spirit means this code https://github.com/NoteCAD/NoteCAD/blob/3e6384e50b2d247a8176f7954d365fee99c30e53/Assets/Code/Constraints/LineCircleDistance.cs#L39

@Evil-Spirit what corner cases? It is simply "center point of circle to line distance = radius".

ayourk, 2026-09-01:
I'm writing as a downstream consumer:
[HobbyCAD](https://github.com/ayourk/hobbycad) drives libslvs through the C
API, and I implemented line-to-circle tangency there because we needed it.

**A patch adding a native `CIRCLE_LINE_TANGENT` is attached**, with the C
program I used to verify it. It takes @ruevs' formulation from earlier in this
thread — one equation, no helper point, no new sketch entity — so it should be
close to what was already agreed here rather than a competing design. The
measurements below are the evidence behind it, and they answer the
over-definition question this seems to have stalled on.

### Why this one belongs in the library

Without it, the tangency composes from three existing constraints, on a helper
point and a helper radius line that live only in the solver system:

```
SLVS_C_PT_ON_CIRCLE     the touch point lies on the circle
SLVS_C_PT_ON_LINE       the touch point lies on the line
SLVS_C_PERPENDICULAR    the radius drawn to it meets the line squarely
```

That works, and it is what @hofmannu was doing by hand in the first post. But
every consumer wanting tangency has to rediscover it, carry two synthetic
entities per constraint, and get the degenerate cases right on their own.

**Plenty of composition is fine, though, and it is worth saying so.** We also
synthesize a collinear constraint — which SolveSpace has nowhere, not in the C
API and not in the application's own constraint types — as `PARALLEL` plus
`PT_ON_LINE`. That one costs nothing: it reuses an endpoint the sketch already
has, adds no entities, and removes the two freedoms you would expect. Nobody
needs a native `COLLINEAR`, and I am not asking for one.

Tangent to a circle is where composition stops being cheap. It is the only
constraint we emit that has to *invent geometry* — a point and a radius line
existing in the solver and nowhere in the document — purely because there is
no point on a circle to hang the equation from. That is the line between
"compose it downstream" and "belongs in the library", and it is the one
constraint I have hit that falls on the far side of it.

**And three equations really are the minimum that way.** A helper point
contributes two freedoms, so any *pair* of equations nets out to zero.
Measured on our build:

```
line + circle, nothing applied                  7 DOF
  + helper point                                9 DOF
  on-circle + on-line                           7 DOF
  all three of the above                        6 DOF
```

So the on-circle + on-line pairing suggested early in the thread constrains
the line/circle relationship by exactly nothing — it picks an intersection
point, which is a secant, not a tangency. The same counting applies to
on-circle + perpendicular. @ruevs' formulation sidesteps all of it: *distance
from the center to the line equals the radius* is one equation with no helper
point, and removes one freedom just the same — one third of the machinery for
the same result. If a native constraint is written, that is the one to write.

### Does it over-constrain?

> When you have Arc-Line incidence at point, and applying Tangent, would you
> expect Overdefinition?

Measured, no:

```
arc + line                                   base  9 DOF
  + coincident (line start = arc start)          -> 7
  + coincident + ARC_LINE_TANGENT                -> 6
[... 67 lines cut ...]

**The measurements of the composed workaround** came from HobbyCAD's own
build, which does carry a few local patches — chiefly a fatal-error handler,
so an assertion reaches the host instead of aborting the process. None touch
solver math or the constraint equations. That handler is incidentally why I
could measure the failing case at all rather than just watch the process die.
I re-ran every figure after rebasing onto `952c11c0` and none moved, including
across b87bfca2.

**The test earns its assertions.** An earlier draft of the equation wrote
`d - r = 0` directly. It compiled, it registered, and it removed a degree of
freedom — and then the solver satisfied it by driving the **radius to -1.5**
with the center 1.5 from the line, because in a workplane
`PointLineDistance()` returns a signed projection rather than a magnitude. The
3D branch does take a magnitude, so this only bites where sketches actually
live. Squaring both sides is what the patch does, and the test asserts the
radius is still positive afterwards, so that regression cannot come back
quietly.

One note for anyone reproducing from a tarball rather than a checkout:
`cmake/GetGitCommitHash.cmake` reads `.git/HEAD` unconditionally, so stock
will not configure without one. A stub `.git/HEAD` gets past it.

[0001-add-circle-line-tangent-constraint.patch](https://github.com/user-attachments/files/31711645/0001-add-circle-line-tangent-constraint.patch)
[circle_line_tangent_test.c](https://github.com/user-attachments/files/31711646/circle_line_tangent_test.c)