solvespace/solvespace #1494: Crash when saving on macOS 15.1

bboydt, 2024-11-10:
### System information

- **SolveSpace version:** 3.1~bb160669
- **Operating system:** macOS 15.1

### Expected behavior

When saving a file using the finder dialog, the app should not crash.

### Actual behavior

The app crashes.
[image]

### Additional information

The issue occurs in `FileDialogImplCocoa::FreezeChoices` at `src/platform/guimac.mm:1432`.

```objc
void FreezeChoices(SettingsRef settings, const std::string &key) override {
        settings->FreezeString("Dialog_" + key + "_Folder",
                               [nsPanel.directoryURL.absoluteString UTF8String]);
}
```

`[nsPanel.directoryURL.absoluteString UTF8String]` is returning `NULL` because `nsPanel.directoryURL` is `nil`.

vespakoen, 2024-12-09:
I cannot reproduce this on macOS, the stack trace showns an issue with the strlen? where do you try to save the file and what filename are you using?

mindingulove, 2024-12-23:
I can confirm it happens when saving, and when importing, no matter which filename is used.

ruevs, 2024-12-24:
@mindingulove does this happen with any model/file or a specific one? If it is a specific one can you attach it here?

commit 3b5f98a7 (2025-02-05): mac: don't crash on file ops without changing dir

Fixes #1494, by only saving the last navigated directory when one actually
exists.
PR #1524 (2025-02-05): mac: don't crash on file ops without changing dir
Fixes #1494, by only saving the last navigated directory when one actually exists.

I'd originally written the following as a reply to that issue, but decided to fix the bug myself instead.

---

I repro'd using 3.1~70bde63c — it happened when saving to the default directory the first few times. After I navigated to a different directory on a subsequent save, the crash no longer occurred; I'd say it's because `Dialog_Sketch_Folder` gets set and so `nsPanel.directoryURL` gets a value when thawed.

To reproduce:

1. Ensure SolveSpace is closed.
2 [... cut]
closed 2025-02-05 by ruevs