solvespace/solvespace #1495: Add a checkbox to "allow skew" of extrusions

phkahler, 2024-11-16:
### System information

- **SolveSpace version:** ALL
- **Operating system:** ALL

### Expected behavior

It should be easy to make skewed extrusions. Perhaps a checkbox in the group text window to allow it.

### Actual behavior

Extrusions groups have 2 hidden perpendicular constraints to lock the extrusion direction perpendicular to the workplane. 

It turns out you can select a workplane different from the one containing a sketch to extrude the sketch, and this will create a skewed extrusion. This is a little awkward because the direction has to be set in the prior group via a little normal vector rather than using constraints in the extrude group.

### Additional information

I have tried commenting out the 2 perpendicular constraint, and it does result in a 3 DoF extrusion. All we'd really need is a checkbox for extrude groups to "allow skew" or similar which would inhibit creation of the 2 hidden constraint. This might be similar to how a constraint can be enabled via checkbox to lock helix pitch.

Is this a good idea?

nobodyinperson, 2024-11-18:
Oh wow if it's that easy it sounds like a really good idea 🤩 I'd love to do skewed extrusions easily without the other finnicky workplane hack.

That change would mean that you can constrain the extrusion axis in the extrude group, right?

With Helix, you can already do skewed extrusions, see here:

[image]

But here you create a 3D line in the same sketch that you want to helix-extrude. I haven't managed to simulate a skewed linear extrude with just helix (worth a try, right? 🙃).

So maybe unifying these operations might be helpful in terms of consistency? That would mean allowing linear extrude also when selecting a point and a line, like with helix extrude. I would have needed that quite often and wondered why something complicated as helix extrude can do it, but a linear extrude can't.

ruevs, 2024-11-18:
@nobodyinperson skewed extrusions are possible already - see here https://github.com/solvespace/solvespace/issues/453#issuecomment-642508274 - this issue proposes a simpler way to achieve it.

phkahler, 2024-11-18:
> My question was whether the proposed solution in this issue will make the extrusion direction be constrainable **after** the extrusion

Yes. That is one of the big advantages. You'd get geometry with 3 degrees of freedom in the new group.

phkahler, 2024-11-18:
> I still think that having Point+Line+`SHIFT+X` extrude like helix extrude would help consistency, but that's a different issue, then, I guess.

That'd be a new variation on extrude. Extrude offsets the original sketch by a vector (x,y,z) which is internally constrained to be perpendicular to the "selected" workplane. Helix, Revlove, and Lathe have the user select a line or normal to use as the **rotation** axis. Helix also allows translation along that same axis while rotating.

ruevs, 2024-11-18:
> @ruevs I am well aware, but going via a manually-placed workplane that's not (easily? at all?) re-orientable or constrainable is hardly a workaround.

Sorry. I did not read your comment carefully.

phkahler, 2024-11-19:
I made a PR #1496 to implement this.

The radio buttons for Extrude gain a 3rd option "skewed".  I'm wondering if this should instead be a separate check box so it can apply to either one or two-sided. I made the internal name ONE_SKEWED in case we want to add TWO_SKEWED and do that.

nobodyinperson, 2024-11-20:
@phkahler This is amazing, thank you! 🤩

[image]

I think, making that `skewed` a checkbox so it can be applied for either one- or two-sided would make sense though.

nobodyinperson, 2024-11-20:
With this something like tapered extrude of a rectangle becomes bit more achievable by intersecting two skewed extrusions:

[image]

phkahler, 2024-11-20:
@nobodyinperson  I really hate the idea of people using this in place of other construction tools that solvespace does not have. But that's to be expected given the situation. You're going to stress the NURBS boolean code with this stuff ;-)

BTW every version of Solvespace prior to 3.0 had problems with skewed surfaces and boolean operations. Turns out the very obscure way of creating skewed extrusions was the only way to create such surfaces. When I first added Helix it was very prone to boolean failures since helical surfaces are both skewed and twisted. Things have gotten better ;-)

ruevs, 2024-11-22:
> You're going to stress the NURBS boolean code with this stuff ;-)

And let the new age of NURBS boolean problem reports begin (this is a good thing) based on https://github.com/solvespace/solvespace/pull/1496 . Let me be the first one ([SkewedCylindersPR1497.zip](https://github.com/user-attachments/files/17873633/SkewedCylindersPR1497.zip)) but this is really a known one.

PR #1496 (2024-11-26): Add skewed subtype for extrusion groups. Update the textwindow radio …
…buttons and equation generation.

Closes #1495
closed 2024-11-26 by phkahler