solvespace/solvespace #1499: Equal Angles Behavior Change

3kakakhel, 2024-12-05:
### System Information

- **SolveSpace version:** 3.1~8c4ac6ee
- **Operating system:** Windows 10

### Expected Behavior

I tried reconstructing one of my tests with the new skewed extrude option. I was able to use the skewed extrude. However, I am not able to make the regular pentagon the way I made it before. See the screenshots below. I made it with a five segment closed polygon onto which I made a set of equal length and angle constraints. It was a simple process of just looping around to apply the constraints; two or three segments at a time. Just change the attached .txt file to a .slvs file.
[image]
[image]
[skew-test-2.txt](https://github.com/user-attachments/files/18016250/skew-test-2.txt)

### Actual Behavior

I tried to use the same process. However, it seems to be stuck on these other angles despite starting out close to the correct angle before applying the constraints. Whereas, before, it was natural to loop around the segments with the equal length and angle constraints. The new ability to apply equal lengths to many segments really helps.

[image]
[image]
[image]
[image]

ruevs, 2024-12-05:
The "g004-sketch-polygon" sketch in your skew-test-2.slvs is not a closed contour:

[image]

This is the reason you need so many equal angle constraints to form a regular pentagon.

A closed contour of five line segments with an equal length needs only two "equal angle" constraints to become a regular pentagon:

[image]

Here is the model from the above screen shot 
[Pentagon.txt](https://github.com/user-attachments/files/18021957/Pentagon.txt)
rename to .slvs

3kakakhel, 2024-12-06:
Thanks for looking into this issue. I am still confused; though, as long as I can make it then it is ok. Your first picture shows it is an open shape. My first picture not only looks closed, but also has the inner coloring of a closed shape. Plus it will extrude the shape (see screenshot). I do not know where something would have changed.

[image]
[skew-test-2.txt](https://github.com/user-attachments/files/18031616/skew-test-2.txt)
(rename as *.slvs)

ruevs, 2024-12-06:
> I am still confused; though, as long as I can make it then it is ok. Your first picture shows it is an open shape. My first picture not only looks closed, but also has the inner coloring of a closed [...]

Look at the check box `relax constraints and dimensions` that is checked in my first screen shot. It temporarily disables enforcement of constraints (except points coincident) to allow you to move/rearrange the model. The contour on your model is indeed closed when all the constraints are "enabled", because they force the top two point to become coincident - but not because they have a coincident constraint. When you check `relax constraints and dimensions` you can move the points and see that they are not constrained coincident. That is how I figured out why your model "tolerates" so many equal angle constraints.

My guess is that when you tried to re-create your model you made a closed contour of line segments when initially drawing them - that would be the logical thing to do.

closed 2024-12-06 by ruevs

phkahler, 2024-12-06:
I had some trouble with this too. After applying a constraint I needed to select it and apply Constrain -> Other Supplementary Angle (u) to get it right. I also did this before making the lengths equal to keep things calm.

ruevs, 2024-12-06:
> After applying a constraint I needed to select it and apply Constrain -> Other Supplementary Angle (u) to get it right.

That is true, but it is not new/changed.

3kakakhel, 2024-12-07:
Thank you, ruevs. I will have to play around to understand that nuance. Does that mean that I have to keep check/unchecking that disable/enable option to make sure my models actually are (hard) closed with a constraint versus (soft) closed by happenstance or by visual effect? Yes, I did just click through to make a loop of line segments when re-trying. I am not sure what or how I did something else the first time.

Thank you, pkhaler. That option should help me when it seems like SolveSpace is, e.g., constraining the outer angle rather than the inner angle. I know it can be effectively equivalent. I still like to be able to choose/set which angle I am constraining.

Seems like it was an issue new to me versus an issue new to the software.

ruevs, 2024-12-07:
> Does that mean that I have to keep check/unchecking that disable/enable option to make sure my models actually are (hard) closed with a constraint versus (soft) closed by happenstance or by visual effect

It does not matter what constraints force a contour closed. The end result is the same - as long as the math works. So your original sketch and mine achieve the same.

The `relax constraints` option is useful when you do not understand why/how some set of constraints work - like I did above.

Its other very useful application is to rearrange a sketch - with already applied constraints - into a different stable configuration/solution - read the tutorial on constraints here https://solvespace.com/constraints.pl and see the example with the pentagon vs. star.

phkahler, 2024-12-08:
@3kakakhel  is should display a message in red when a sketch is not a closed contour - that may be an option in the configuration menu.

It will also color in the inside area once it's closed, but the default color for that isn't very visible on many displays. You can change it though.

3kakakhel, 2024-12-08:
ruevs, thank you. I like how that constraint guide made the polygon. 

phkahler, thank you. I have seen both of those effects.