solvespace/solvespace #1507: Trackpad gestures on Linux

polyfloyd, 2024-12-16:
I use Solvespace exclusively on my laptop which has a touchpad. The current controls are sufficient to model with, but the right trackpad button is does not feel nice to use.

It would be nice to also support zoom/pan gestures on the Linux/GTK based platform.

Related, gestures on Mac: #68

I would be happy to spend some time on this myself. However, I am wondering whether it would be better to first finish the port to GTK4 before working on this :thinking:

phkahler, 2025-08-25:
> I would be happy to spend some time on this myself. However, I am wondering whether it would be better to first finish the port to GTK4 before working on this 🤔

@polyfloyd  To be clear, I don't see myself leading this thing forever. But my vision for the project is GTK4 with single window (see the web version) on all platforms. I've tried the GTK4 port myself and it's outside my area of expertise so it'll be up to others to make it happen. Having better trackpad functionality would be great in the present, and might also carry over to GTK4. No need to wait IMHO.

Personally, I want the arrow keys to do rotations as well. AFAICT they're not handled the same as other keyboard inputs and that might also be platform dependent. Just something to consider when improving the laptop experience.

polyfloyd, 2025-09-09:
I'm a little busier these days, but maybe I will have some time to look into this still.

@phkahler can your GTK4 attempt be found online so I can perhaps start from there?

phkahler, 2025-09-10:
> I'm a little busier these days, but maybe I will have some time to look into this still.
> 
> [@phkahler](https://github.com/phkahler) can your GTK4 attempt be found online so I can perhaps start from there?

Not very useful but you can start with my [GTK4 branch](https://github.com/phkahler/solvespace/tree/gtk4).  I would only take the first 2 commits though, which just do basic changes - bumping the C++ version used (GTK4 requires newer than current solvespace) and change the includes, make a new .cpp file.

I would ignore the later commits because I view the attempt as a complete failure. A better approach might be to make stub functions out of everything in that main file, and then start filling it in. Window creation would be the first thing to make work, followed by I suppose getting menus to populate.

Solvespace has an abstraction layer between it and the GUI toolkit, so there are functions that go both ways. It has it's own "event" objects that get passed to solvespace for example - populated from whatever the chosen toolkit provides. It also has functions to build menus, which need to call whatever the toolkit requires for that - and object wrappers I think, so there is a solvespace menu class but it'll have to wrap a GTK4 object of some kind.

It can't be too hard - it's only 1600 LoC. But by the time you're done you will understand both sides of the abstraction. I don't think it can be done by renaming API calls to their new versions blindly, which I what I tried to do. Like google how to create menus in GTK4 programatically - the old menus are gone, but there is something in their place.