solvespace/solvespace #1508: Unsolvable Constraints

eternal-sorrow, 2024-12-27:
### System information

- **SolveSpace version:** 3.1~flatpak
- **Operating system:** Gentoo Linux
### Expected behavior

Very simple case. Draw a circle, constrain the circle centre to be at the origin, constrain the circle diameter to 3mm. Extrude the sketch, make the extrusion two-sided, constrain the extrusion length to 10mm. Nothing too complicated, should be working, right? Am I doing something wrong? I'm a complete noob at SolveSpace an CAD. Is there a better way to make a cylinder 3x10mm?

### Actual behavior

Unsolvable Constraints.
[image]
[image]

### Additional information
[cylinder.slvs.txt](https://github.com/user-attachments/files/18262294/cylinder.slvs.txt)

mkyral, 2024-12-27:
I'm not sure what exactly you did, but I was able to fix it only by deleting the extrusion layer and do it again.

[image]

[cylinder_fix.slvs.txt](https://github.com/user-attachments/files/18263295/cylinder_fix.slvs.txt)

ruevs, 2024-12-28:
@eternal-sorrow  the problem in your model is that the constraint setting the length of the extrusion is in the XY plane (it was added when the mode was `Sketch | In Workplane`) - this means that the two end points of the extrusion need to be 10mm apart in the XY plane - which is impossible because the extrusion is perpendicular to the XY plane and therefore the projections of the two pints in the plane coincide.

The constraint needs to be added when the mode is `Sketch | Anywhere In 3d`

See further information se the reference:
https://solvespace.com/ref.pl#Constraints

[image]

closed 2024-12-28 by ruevs