solvespace/solvespace #1527: Request: completely free objects

bkauler, 2025-02-16:
I am tantalizingly close to creating an object that is completely free to move in any direction, free of the coordinates. I did this by starting sketch in 3D and create a triangle, then used the triangle as the workplane, then created a sphere then two tubes, all locked to the same central point.

The idea is to create a "ball joint" that is completely free to move in any direction. I am making use of the new "skew" feature of the tube extrusions.

It works great, except for this:

[image]

I need a feature enhancement to set the angle of the tube outer face to zero degrees. That is, to set the angle of the normal at face of the tube to not be skewed.

Here is my test file, renamed with a false ".gz":

[test-joint-1.slvs.gz](https://github.com/user-attachments/files/18813477/test-joint-1.slvs.gz)

Is this request madness? I can see huge potential for objects that are totally free to move anywhere in 3D, and can be joined together.

The requested enhancement would be, I think, quite easy to implement, for you guys who know the code how that skewed feature is implemented.

bkauler, 2025-02-16:
In fact, I think that my requested enhancement will be needed by everyone who ticks the "skew" checkbox. See this:

[image]

There is a central post and a tube sticking out, not perpendicular. I do not want the outer face of that tube to be skewed!

ruevs, 2025-02-16:
I would bet (99%) - it is not going to happen. You will have to wait for loft (https://github.com/solvespace/solvespace/issues/439). When/if it gets implemented some day.

As an annoying and clunky workaround - cut off the end perpendicular to the extrusion direction with an extra difference operation.

phkahler, 2025-02-16:
Skewed extrusion works exactly like it's supposed to. This is not going to change. It is meant to create skewed extrusions, not to change their orientation or pivot them as a part.

> I do not want the outer face of that tube to be skewed!

Then you should not be using the skewed extrusion option ;-)  You need to make the 2d sketch perpendicular to the axis you'd like to extrude along.

Also, what happened here? I thought the original request was something completely different when I saw it on my phone. Did that get deleted? I thought you wanted parts that can be independently positioned and oriented in a sketch. In that case, you'd want to create an assembly. You can read the section of the manual for assembly here:
[Link / Assemble](https://solvespace.com/ref.pl#Link)

If you make the round post a separate file ( just an extruded circle ) then you can link it in this file and position or orient it however you like and its shape will remain constant. You can keep it a separate part "link as assembly" or fuse it with the parent part using the "union" option.

closed 2025-02-16 by ruevs