solvespace/solvespace #1532: Closed polygon drawn in sketch workplane looks normal unless looked from an angle, lines disconnected, dots at random height

fedd, 2025-02-22:
Is it a known bug or am I missing something?
I drew a closed polygonal line in Sketch in workplane mode and it looked okay but when I changed the view angle it turned out to be completely broken, can't imagine how it can be extruded or something.

Ortho view:

[image]

Isometric view - surprise!

[image]

Discotheque:

[image]

Have I lost my project?

ruevs, 2025-02-22:
@fedd You turned on the "exploded view" feature introduced in version [3.1](https://github.com/solvespace/solvespace/blob/master/CHANGELOG.md#31). Press the `\` key to turn it off. You can do the same from the menu `Virew -> Show Exploded View`.

At the bottom of the `view` configuration screen (click from the "Proroperty Browser") there is an option `explode distance` that controls how far apart the items are when the exploded view is on.

https://github.com/solvespace/solvespace/pull/1003
https://github.com/solvespace/solvespace/issues/173
https://github.com/solvespace/solvespace/issues/1225

closed 2025-02-22 by ruevs

fedd, 2025-02-22:
I see, thanks!!