solvespace/solvespace #1539: macOS build instructions in README don't work

ad-si, 2025-03-02:
OS: macOS Sequoia 15.3.1 arm64

```txt
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=ON
-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- The ASM compiler identification is AppleClang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
Call Stack (most recent call first):
  /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/share/cmake/Modules/FindOpenMP.cmake:616 (find_package_handle_standard_args)
  CMakeLists.txt:123 (find_package)

-- Configuring incomplete, errors occurred!

$ cmake .. -G Xcode
CMake Error: Error: generator : Xcode
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
```

ruevs, 2025-03-02:
Just build it without `-DENABLE_OPENMP=ON`
As to why it is not being found - even after you "brew-installed.it" -  @vespakoen or @dgramop may be able to help you. They are on macOS.

ad-si, 2025-03-02:
Thanks, that helped, but now it fails during build:

[errors.txt](https://github.com/user-attachments/files/19043748/errors.txt)

iscgar, 2025-03-03:
You're likely running into [this issue](https://github.com/godotengine/godot/issues/101035), and need to apply the following patch (taken from [here](https://github.com/godotengine/godot/pull/101040/files#diff-1c411c43a7c94467aef4a0720d1f4a3ad57df374de5b8b1607633be0f031191e)):

```diff
diff --git a/extlib/freetype/src/gzip/ftzconf.h b/extlib/freetype/src/gzip/ftzconf.h
index 3abf0ba03b..d88a82a2ee 100644
--- a/thirdparty/freetype/src/gzip/ftzconf.h
+++ b/thirdparty/freetype/src/gzip/ftzconf.h
@@ -215,7 +215,7 @@
 #   define FAR
 #endif

-#if !defined(MACOS) && !defined(TARGET_OS_MAC)
+#if !defined(__MACTYPES__)
 typedef unsigned char  Byte;  /* 8 bits */
 #endif
 typedef unsigned int   uInt;  /* 16 bits or more */
```

iscgar, 2025-03-03:
Ideally FreeType should just be updated. This specific fix first appeared in zlib 1.2.0 (https://github.com/madler/zlib/commit/7c2a874e50b871d04fbd19501f7b42cff55e5abc, 14 years ago), and adopted by FreeType as part of version 2.12.0 (https://gitlab.freedesktop.org/freetype/freetype/-/commit/a25e85ed95dc855e42e6bb55138e27d362c5ea1e, 3 years ago).

However, if the maintainers prefer to avoid such a large change, then this specific issue should indeed probably be fixed.

ruevs, 2025-03-03:
Updating FreeType should be fine. I'll do it when I'm on a computer.

iscgar, 2025-03-06:
I'm assuming you didn't have time to look into it yet, so I went ahead and opened #1543.

ruevs, 2025-03-07:
> I'm assuming you didn't have time to look into it yet, so I went ahead and opened [#1543](https://github.com/solvespace/solvespace/pull/1543) to fix it.

Thank you @iscgar I did it locally but had not pushed and opened a pull request.

ruevs, 2025-03-07:
@ad-si  Please check if the latest master compiles without the [patch](https://github.com/solvespace/solvespace/issues/1539#issuecomment-2694149011) on macOS. The problem should be fixed because we updated to the latest FreeType here https://github.com/solvespace/solvespace/pull/1543 (remember to update your submodules).

closed 2025-03-07 by ruevs

ad-si, 2025-03-09:
Yes, that fixes it! Thanks for your efforts! 😊

PR #1543 (2025-03-07): Update the vendored FreeType version to 2.13.3
The existing version (2.7.1) is using an ancient zlib version, which fails to build with modern Xcode tools on macOS.

Fixes #1539
commit bd7b6278 (2025-03-07): freetype: update the vendored version to 2.13.3

The existing version (2.7.1) is using an ancient zlib version, which
fails to build with modern Xcode tools on macOS.

Fixes #1539