solvespace/solvespace #1548: Dxf-imported lines cannot be mixed with user-added lines because of style

dgramop, 2025-03-24:
BLUF: Lines imported from DXF have a different "style" than lines generated by the user. If the user tries to create lines to, say complete a DXF shape, solve space doesn't let the user extrude because they're not the same style.

Noticed on master: (bd7b6278)

I imported a DXF generated from an SVG of a NACA airfoil. The generated airfoil wasn't a closed shape, so I connected the two end points. Solvespace said "not closed shape or not all same style". After some fiddling around, I realized I could delete all the lines and replace them with user created lines to get a closed shape.

I went into the savefile and deleted all the references to the dxf style. Then, I was able to close the shape, and extrude it.

I supposed we delete

dgramop, 2025-03-24:
The function in src/importdxf.cpp looks like it has a quick fix - deleting the line that gives these entities their own style. 

If we don't want this fix, maybe we have to support mixing styles of lines in other places in solve space?

phkahler, 2025-03-24:
@ruevs what about an option to give them their own style vs regular?

Also, can it link dxf files?

ruevs, 2025-03-27:
Entities imported from [DXFs](https://free-dxf.com/design-nautical-compass-north-arrow/free-download-dxf-cdr-ai-svg/UQaW6) do get their own style and I believe this was intentional. What is the problem with just setting them to the default if one wants them to "participate" in solid objects?

1. `Ctrl-A`
2. Right-click
3. `Assign to Style | No Style`