solvespace/solvespace #1551: Transparent icon doesn't work well on macOS.

7ombie, 2025-03-28:
While macOS allows icons to use the alpha-channel to form icons of any shape, it also requires that users click on the opaque part of the icon. The SolveSpace icon is almost entirely transparent, making it difficult to click on.

I only realized this is how macOS works when I kept clicking on the SolveSpace icon and nothing happened. Every other icon has an obvious enough opaque area that I just never missed an icon before.

There basically needs to be a shape in the background.

[image]

phkahler, 2025-03-28:
A black background might be nice. Maybe rounded corners?

7ombie, 2025-03-29:
Or maybe a Cartesian plane, with dotted edges.

phkahler, 2025-03-29:
@vespakoen  what art asset is used for the Mac icon?

iscgar, 2025-03-29:
@phkahler all of the PNG files under `res/cocoa/AppIcon.iconset`.

stllfe, 2026-03-17:
Hello there! I'm a big fan of SolveSpace, as it's the only CAD program out of all the ones I've tried that I immediately got the hang of. 

I have experience in graphic design, so if you're interested, I'd be glad to draw a new, more modern icon.

I ran a quick brainstorming session with a Gemini model and generated several draft concepts:

The one I like the most:
[image]

This one is a bit too much, but I like the arrow idea:
[image]

This one tries to stay closer to the original neon-like colors:
[image]

And this one is just to show off how cool LLMs can be used as a rendering engine lol:
[image]

---

If we settle on an icon variant, I can redraw it from scratch in Figma and provide all the necessary source files.

7ombie, 2026-03-20:
They do look nice though.