solvespace/solvespace #1569: Crash with/without "Handle isn't unique" error message

blinkenlight, 2025-04-17:
Solvespace version: 3.1~bde63c (openMP, 32-bit)
OS: Windows 7
Sample file: well, it's complicated. It's kinda large...

I have a moderately complex file of assemblies that teeters precariously on the edge of crashing. Change the number of translated copies up by a few? Crash. Import ANY new assembly? Crash. You get the idea. Most of the time there is NO error message at all whatsoever - only a "ding" then I have to kill Solvespace form the task manager. But sometimes there is, and this is what it says:

```
File D:\a\solvespace\solvespace\src\dsc.h, line 495, function
SolveSpace::IdList <class Solvespace::Entity, class
SolveSpace::hEntity>::Add:
Assertion failed: FindByIdNoOops(t->h) == nullptr.
Message: Handle isn't unique.

Generate debug report?
```

This seems to be an ancient issue if I'm not mistaken, as mentioned in [this old post](https://github.com/solvespace/solvespace/issues/932#issuecomment-808785294) - some kind of 16-bit resource depletion, apparently. While it's nice that we all know about it, this creates a huge problem for me, being unable to continue doing, well, basically ANYTHING with my design, even after enduring the obligatory seconds-per-frame punishment at each action. Please advise, what do? Any workarounds...? Heavens forbid, a patch...?

ruevs, 2025-04-18:
This is indeed a known limitation.
https://github.com/solvespace/solvespace/issues/932#issuecomment-808785294
https://github.com/solvespace/solvespace/issues/1227

Removing it requires a change in the file format, and all the hurdles related to keeping backward compatibility with old files and converting between the two.

Machines, have gotten faster, we've made SolveSpace much faster and these days it is possible to hit this limitation...

blinkenlight, 2025-04-26:
...so is the conclusion simply "too hard, fuggedaboutit in any kind of future that could be considered near"...? Sigh. I suppose it's true that the only thing preventing Solvespace from being the best 3D CAD there is and conquering the world is... Solvespace.

iscgar, 2025-05-01:
> ...so is the conclusion simply "too hard, fuggedaboutit in any kind of future that could be considered near"...? Sigh.

I'm not a maintainer, and I sympathise with your frustration, but let me say this: no, it does not mean that at all. Lifting this limitation is not easy, and it requires time and effort to come up with a design that works, doesn't break with edge cases lurking in dark corners of the code base, and as @ruevs mentioned is able to work with old files. Menawhile, there aren't many people who contribute to SolveSpace these days (let alone regularly work on it), and those who do have less free time on their hands than they used to. They also have their own pet peaves that they'd like to tackle, so they prefer to use their free time to work on them rather than on what you may consider the most pressing issue.

The reality is that while it is an issue that had come up in the past, most users find ways around it (by e.g. linking multiple files for their design), so it's probably not the first thing that comes to mind when people think about ways to improve SolveSpace. However, if you have a patch ready, then by all means, please submit it for the benefit of the SolveSpace community. If not, perhaps you'd like to try your hand at solving the issue and becoming a contributor? I can't speak for the maintainers, but I personally would also love to see this limitation lifted at some point, or work on it myself when I find the time if no one else beats me to it.

blinkenlight, 2025-05-02:
> The reality is that while it is an issue that had come up in the past, most users find ways around it (by e.g. linking multiple files for their design)

While I do think a fix for this can't come soon enough, I'd be quite happy with a functional workaround too; however, yours is not one - my design is a simple plastic box with some electronics inside, NO sketched entities whatsoever, only nine other designs loaded as an assembly each in its own group. Then I translate the last one four times in a new group, and after that, Solvespace immediately crashes each time I try adding ANY further assemblies. It also crashes if I just try increasing the translate count over 15 or so. Yes, some of those assemblies are themselves assembled from other sub-assemblies; I don't think that's unfair. The design is also early-constrained to triangles, due to Solvespace's inability to handle NURBS for more than 5 minutes - this might contribute in some way as well, no idea.

But in the end, assembling is not a workaround at all for this, as much as I'd love to see a working one. I don't think this is excessive complexity that cannot be fairly expected to NOT crash a functional CAD software, especially since all I'm doing is bringing in "inert" models and immediately constraining them to something else fully - nothing can possibly move in there, there should be nothing to recalculate except the graphic representation to be drawn itself.

> However, if you have a patch ready, then by all means, please submit it for the benefit of the SolveSpace community.

I would enjoy that greatly, and since I'm not a psychopath I would obviously not hold one back if I had it. Unfortunately, due to certain health issues my available spare time and energy is literally negative, even the simplest tasks at home ending up being multi-month projects if I manage to ever complete them at all. Not fishing for sympathy here, just stating some facts - if I could work this out, I WOULD. The other thing is my domain is embedded coding, not x86, and C++ looks halfway like hieroglyphs to me; tried changing that, repeatedly, didn't work. I'd much rather handle ten thousand lines of C (or even assembly) than ten of C++.

> In my head I've had an idea/plan how to do it for a while

I appreciate the intent. I'm not sure it's of much help to me personally for the reasons outlined above, but it's great to even see a plan outlined. The "2^32 entities in a group" does baffle me though - the limit most certainly doesn't seem to be triggered by anything inside ONE single group. As for the memory requirements - as much as I'm a staunch believer in backwards compatibility, by now 32-bit versions have long been quasi-universally regarded as unfeasible to support, especially anywhere they actually bump up against the 32-bit memory limit. Four gigs of RAM long stopped being a workable amount of memory on anything that has a GUI, and I say that as someone who works on such a machine daily (don't ask). As much as I'm all for preserving support for anything that practically can be, I honestly believe that in 2025 a 3D CAD program needs not be apologetic about not working all that well on a 32-bit machine, if at all.

blinkenlight, 2025-05-02:
@phkahler No problem, there's nothing confidential about it - the only reason I didn't attach it outright was size considerations. Thankfully, it seems to zip down to a somewhat manageable size after all. The root design is "buzzybox.slvs", I included the assembled ones as well.

[Box.zip](https://github.com/user-attachments/files/20018200/Box.zip)

ruevs, 2025-05-02:
> The other thing is my domain is embedded coding

You; and [the original author of SolveSpace](https://cq.cx/index.pll); and the current maintainer; and [the previous maintainer](https://github.com/GlasgowEmbedded/glasgow); and I; and (I strongly suspect) a few of the other significant contributors over time. There is a pattern :-) I think the reason is that people who can program and know of SolveSpace found it and liked it because they wanted to draw something "quick and dirty" - geeks with 3d printers, electronics tinkerers etc. - often these people are embedded programmers.

Mechanical engineers either have/use something "professional", or can not program.

I am not sure why we do not have more 3d graphics/game developers - probably they contribute to Blender and/or Plasticity. SolveSpace is too restrictive for them. @rpavlik is an important exception to this :-)

ruevs, 2025-05-02:
> The "2^32 entities in a group" does baffle me though - the limit most certainly doesn't seem to be triggered by anything inside ONE single group.

I wrote it as a joke.

Joke aside; very roughly speaking: There are a few "pahs" to the "handle isn't unique" error - if/when we make handles 64bit, 32 of those bits will be used for handles within a group (long story, if you are curious either look in the code or here #1574). At that point that would be the only "feasible" way to cause the error, because no one is overflowing unit64 with objects in a CAD software any time soon.

Thank you for posting the model, I'll also take a look at it when I'm on a computer.

iscgar, 2025-05-04:
Adding some debugging prints, this is what I get when loading the model:

```
96 params
98020 entities
12 groups
g001: 0 entities
g002: 0 entities
g003: 1390 entities
g004: 2529 entities
g005: 908 entities
g006: 908 entities
g007: 908 entities
g008: 908 entities
g009: 908 entities
g00a: 783 entities
g00b: 17773 entities
g00c: 71076 entities
```

`g00c` is the translate group, and it already overflows the amount of allowed entities (~71k when the limit is 65536). With the way handle IDs are incremented, this effectively overflows into the next group's handle domain, which is why any group you add after this will clash with its entities. The only way I can see out of this is by doing the translate in two steps, each one repeating at most 3 times (3 * 17773 = 53319, which is still under the limit).

blinkenlight, 2025-05-04:
@iscgar Thanks for taking a look. I arrived at a somewhat similar conclusion blindly - replaced the translation with four separate instances and I could now load a new assembly afterwards. Unfortunately, this is still not a solution - I started throwing in assemblies en-gros after this just to see what happens and sure enough I started crashing after another nine or so more (smaller) assemblies even when using four separate "17773" sub-assemblies. I can't even tell if it's the same cause or not, because there's no error message, and this "handle" bug doesn't throw one either 90% of the time - so it might be the same error or it might not be.

iscgar, 2025-05-04:
Can you describe the exact steps that you're taking so I could try to reproduce it and diagnose what's happening there? (I assume it's still the same model, and you're just linking multiple copies of existing assemblies. If not, and you have no problem sharing, I would appreciate it if you could attach the new model)

ruevs, 2025-05-04:
> because there's no error message, and this "handle" bug doesn't throw one either 90% of the time - so it might be the same error or it might not be.

In fact it does always throw an error message. But on Windows (10, 11) the error dialog for some reason ends up behind the other windows (and is thua invisible) and the main SolveSpace window just spears "hanging" - you can ALT+Tab to the error message or if that does not work press Windows+M to minimize all other windows.

By the way your model is really nice and I see why this limitation frustrates you.

blinkenlight, 2025-05-05:
> Can you describe the exact steps that you're taking so I could try to reproduce it and diagnose what's happening there?

I just replaced the four-times translation with four separate assemblies of the same thing (which is where the model attached below stands - just unzip it into the same folder as the earlier one), then start adding more simple assemblies (such as one of the pushbuttons) again and again, without doing anything else, until the crash happens. It always takes less than ten buttons, but it's far less deterministic than the OP bug. The first clue that you're getting close is that the "edges of solid model" white contour suddenly goes missing - on the next "link/assemble" attempt there's usually a crash. But this one doesn't throw an error message or a "ding", the window just closes and I get the Windows message about the program having crashed. So yeah, my mistake, that seems to be something else - or at least a different symptom.

> In fact it does always throw an error message.

Well, we're both right - I was quite sure there was no error message at all on most cases, not just hiding under other windows (I have been looking for it exactly as you described); it turns out that's because in those cases the error message only appears for me (as a window I can alt-tab to) well over a minute later, which most times I did not wait for. Before that, it simply doesn't exist in the alt-tab list of windows. In "normal" cases the message shows immediately, that's when I was seeing it (even if popping behind something else). I'm on Win7 btw, not 10/11.

> By the way your model is really nice and I see why this limitation frustrates you.

Thanks! :)

[Box2.zip](https://github.com/user-attachments/files/20042726/Box2.zip)

iscgar, 2025-05-06:
> I just replaced the four-times translation with four separate assemblies of the same thing (which is where the model attached below stands - just unzip it into the same folder as the earlier one), [...]

I forgot how slow the 32-bit version is. I feel your frustration regarding the perfromance of SolveSpace with your model, and I really admire your patience.

So the issue in this case is simply memory exhaustion. After the ninth push button is linked, SolveSpace is at around 2.6GiB on my machine before running out of memory and aborting. Curiously though, if I load the savefile after the crash, SolveSpace is sitting at ~800MiB, and I can add more assemblies, which likely indicates that we're keeping in memory a bunch of garbage objects that we shouldn't keep (this is something that I've also encountered with my `List` and `IdList` work, which saw a reduction of more than 50% in peak memory usage on some models compared to master), so there's definitely some work that we could do to improve the situation, but the best course of action would be to simply build and use a 64-bit version from source instead, as It's much faster (due to the available registers and better CPU instructions), and it can use much more memory.

phkahler, 2025-05-07:
@blinkenlight  Since you're doing electronic boxes, have you tried linking IDF files? Currently they only pull in the board, and there may be some issues there. It's a feature I added over a few days and haven't really polished because I'm not sure many people actually care about it. If some folks are actually going to use it I might add code for the limited part models of IDF. I think the crude part models are why nobody seems to use IDF in practice, even in industry.

blinkenlight, 2025-05-07:
@phkahler Honestly, I've never even heard about IDF files before - but now that you've mentioned them I quite like how they allow importing the PCB back into Solvespace, with attachment points; looks like I'll be using them, any further progress would be of course welcome. Obligatory side note: there seems to be a classic "which arc" bug (had to fight about a million of them in GCAM SE as well) somewhere in IDF handling, see the small PCB from one of my earlier boxes attached below.

Normally, I explore general enclosure fit / layout options in Solvespace and once the design is done I export the PCB outline to KiCAD for doing the actual PCB layout. After that, I can take the Gerbers and the Solvespace dxf/svg's into GCAM SE and isolation-mill out the PCB and the enclosure holes on my small desktop CNC, including any engraved faceplates if applicable. The Solvespace model is not just a pretty picture, I'm doing the actual fabbing based on it.

[Snap Timer PCB.zip](https://github.com/user-attachments/files/20092746/Snap.Timer.PCB.zip)

phkahler, 2025-05-09:
Another though about this issue that could use some investigation.  Just expanding the handle size to allow more than 64K per group will probably lead to complaints about solvespace being slow on large models (rather than crashing on them).  A lot of places in the code scan the entire list of entities looking for some (or one) that meet a particular criteria. Perhaps the most common thing is to look for ones that belong to a particular group. This scanning will just slow down with larger models.

What if each group had a list of entity handles that belong to the group? This list could be created as entites are created or read from a file. It would be more bookkeeping but would eliminate the need to scan the whole list. Each entity can (does?) contain a handle for its group, and at that point there would be no need to store the group in the upper half of the handles. The handles could then be expanded to 32bits, and existing files might be compatible just treating the existing handles as a single 32bit value.

group::generate()  looks through entities for ones belonging to opA. Imagine if we just passed in the list of entities from that group directly.

There is a bunch overlooked here. Like what about entities that are from a request? Maybe they're still added to the group entity list and they contain their request handle as a member?

Storing information in the handle is usually a design error, but I think it was done here to avoid looking all over memory in order to do these kind of queries efficiently.

iscgar, 2025-05-10:
> What if each group had a list of entity handles that belong to the group? This list could be created as entites are created or read from a file. It would be more bookkeeping but would eliminate the [...]

Each group already holds a list of its entities using `remap`, so they can be looked up directly if we wanted to (you just need to construct the final handle by calling `hGroup::entity()` on the group's handle and passing each value from the `remap` table as an argument). Note that we won't always find an entity this way, because the `remap` table is never trimmed down when group entitites are removed (by e.g. reducing the amount of copies for a translate group), and that's something that we probably want to fix anyway since currently it needlessly bloats the size of the file.

The code in `Group::Generate()` seems to be trading off the complexity of sequential iteration against binary searching the entities `IdList` for each entity, but it's wrong for two reasons. First, it blindly extracts the group handle from the entity handle, without checking first if it's an entity that was generated by a group or a request. And second, sequential iteration like that instead of binary searching each entity only works efficiently for models with few groups and a small number of entitites per group. Since group entities are only entities that are directly created by the group (using `remap`), we can directly look them up as I explained, but we could make it even faster by using `std::lower_bound()` to get an iterator to the first group entity, and then we could just use the returned iterator to sequentially iterate until we encounter an entity whose parent group handle isn't the one we're looking for.

> There is a bunch overlooked here. Like what about entities that are from a request? Maybe they're still added to the group entity list and they contain their request handle as a member?
>
> Storing [...]

The code in `Group::Generate()` is only concerned with entities directly generated by the group through `remap`, not with entities generated by requests (this is evident from the fact that the iteration code that you mentioned only exists for extrude, lathe, revolve, helix, translate, and rotate -- all groups which generate their entities through `remap`).

But in any case, iterating request entities could be done just as efficiently -- you simply use the group's request list, and for each request you use `std::lower_bound()` (passing the result of calling `hRequest::entity()` on the request's handle in this case) and then iterate until you hit an entity that has a different request handle as a parent.

phkahler, 2026-01-07:
> But in the end, assembling is not a workaround at all for this, as much as I'd love to see a working one. I don't think this is excessive complexity that cannot be fairly expected to NOT crash a [...]

@blinkenlight if you're still here, I reread this and realized some things. When you link another slvs file, it does bring in all the entities in that sketch into one group so that they can be used in constraints. There aren't free parameters for the solver for each entity, the group as a whole has like 7 parameters (4 orientation and 3 position) but the entity count is still large. Also when you link an STL file you'd think there are no entities, but Solvespace will assign an entity where there are "sharp" corners for some arbitrary definition of "sharp" that is in the code. It will limit the total number of points but IIRC the limit is in the thousands, so creating a repeat group will multiply that quickly and could hit the limit - something I never thought about at the time.

Also wanted to respond to your other comment:  "Sigh. I suppose it's true that the only thing preventing Solvespace from being the best 3D CAD there is and conquering the world is... Solvespace."  This makes me sad because it's so true. It's also why I'm still working on this in my spare time. I really like Solvespace and *want* it to be the tool that it looks like.  Version 3.2 will be light years beyond what 2.3 was, and yet it's still not where it needs to be.

rpavlik, 2026-01-07:
Ah, I may be a 3d developer... but I come from having done CAD constraint stuff during my PhD, and I was doing a fair bit of hardware/embedded work when I did my largest SolveSpace contributions :) I'm still pretty systems-y and low level much of the time.

However, I have been burned by attempting to simplify/improve IdList and its peers in the past, so I'll opt out of doing things on this issue now. I will say, a few years ago it would have gotten unbearably slow long before you hit this limit (which is why I originally contributed), improvement here is a matter of incrementally dealing with the next most painful blocker.

Just as a reminder, because it's easy to lose sight of this: I, and many others, do actually use SolveSpace as-is with great success. (Heck I use 3.1 builds still.) Certainly there is an element of adapting to the tool (think ahead about fillets!), as well as the fact that I've spent far more time with SolveSpace than any other MCAD package. It is a peculiarity of the user experience field that every user assumes their mode of usage and their experience is typical, when it is almost certain nobody consciously thinking about the user experience is a typical user. I was once the person with the allegedly-simple-but-limit-hitting models myself.

Conquering the world need not be the goal, this is something useful for ourselves and many others. If folks can find the time and talents to address some of the remaining limitations, increasing the number of people that this is useful for, great! But there is not necessarily any motive (profit or otherwise) to 'increase market share', at least as I see it.

blinkenlight, 2026-01-08:
@phkahler oh I'm definitely around... :) Thanks for the insight, it does help putting things into perspective. And of course, I'm much looking forward to any new release. Really wish I had the spare energy, time and coding skill to tear right into this myself, but alas...

@rpavlik for me it's not so much about some proverbial Linux-style "THIS will finally be the year of Solvespace!" popularity push, it's more like wistfully gazing at a beloved tool in your hands going "you're SO close of being able to do what I need to get done..."