solvespace/solvespace #1608: Boolean difference error

stesj4, 2025-07-10:
I'm trying to cut an arc tangent object from a cylinder.

[image]

[image]

using version 3.2 (3.2~960f08b3) on a Mac with Sequoia 15.5

phkahler, 2025-07-11:
This doesn't quite fit the specific issues I'm aware of. It fails when the difference extrusion gets past a certain length. It has rounded corners, and it fails when the seam between the curved surface and the bottom flat face goes outside the cylinder.

It does not fail if I replace the arcs with straight lines, so it seems related to the tangent surfaces of the second extrusion. But it's not a case of 3 surfaces tangent along a line, which is one of the most prominent bugs. Reminds me of [315,](https://github.com/solvespace/solvespace/issues/315) but that's fixed unless there's a bug in the fix.

stesj4, 2025-07-12:
I was trying out this exercise:

[image]

wolfpackmars2, 2025-07-13:
using 1608_NURBS.slvs I can restrict the difference cut a length of 9.94987. It falls apart once the cut reaches a length of 9.94988.

I also tried a couple of different approaches but was unsuccessful. If it helps, I attach my test files. They should be helpful as additional test files when a solution is found.

[1608_NURBS_alttestcase.zip](https://github.com/user-attachments/files/21202278/1608_NURBS_alttestcase.zip)

wolfpackmars2, 2025-07-13:
I believe the problem is caused by function MakeCopyTrimAgainst in boolean.cpp

https://github.com/solvespace/solvespace/blob/960f08b34ca5f5db30c3981d1708897bac156c35/src/srf/boolean.cpp#L488

This code generates the error that gets thrown when the problem occurs:
https://github.com/solvespace/solvespace/blob/960f08b34ca5f5db30c3981d1708897bac156c35/src/srf/boolean.cpp#L681
```    if(!final.AssemblePolygon(&poly, NULL, /*keepDir=*/true))
#pragma omp critical
    {
        into->booleanFailed = true;
        dbp("failed: I=%d, avoid=%d", I+dbg_index, choosing.l.n);
        DEBUGEDGELIST(&final, &ret);
    }
```