solvespace/solvespace #1609: union/assemble radio button in the extrude section

stesj4, 2025-07-14:
Just a question that have been puzzled me for a while...
What's the difference between the "union" and "assemble" radio buttons in the extrude section?
I use the assemble choice when the union option fails, which happens quite often.
Why not remove the union option all together?

The triangle mesh change a little bit, but not in the inside of the model, so it looks like the assemble option also makes an union.

I use Solvespace for my 3d printer, not for mechanical moving machineries.

(Sorry if this has been answered in other threads)

:)

phkahler, 2025-07-14:
When you use "assemble" there is no change to the geometry at all.  If you stack 2 cubes you will have a model with 2 cubes - the touching faces will still be present.  When you use "union" the touching faces will be removed and the coplanar faces on the sides will get fused together and you'll have a single rectangular object twice the height of one of the cubes.

Assemble was intended for assemblies where you want parts to remain separate even if they may be touching. When you create an assembly you can check for interference of the parts under the "Analyze" menu. If you make the group union they will just be combined.

I think assemble was added as an option for non-assembly groups because it's faster (because it does nothing), and perhaps to avoid bugs with unions.

TLDR:  Assemble does not actually combine things.  Union fuses two objects into one.

stesj4, 2025-07-15:
That's how I can see that it looks like no inner hidden faces. But....
When I open the "assemble" stl file in Blender there are inner faces/mesh between the joined objects.
In the "union" stl file there is no inner mesh.