solvespace/solvespace #1625: Can't drag points on macOS Tahoe

lukexi, 2025-09-16:
Hi, noticed this during the Tahoe beta and it seems to still be true in today's released version of Tahoe (on both SolveSpace 3.1 and 3.2RC1) that dragging points results in the points only moving a few pixels.

(I saw someone reported this in the forum but didn't see any open issue for it yet)

ELadrimonos, 2025-09-16:
Have the same issue

ruevs, 2025-09-26:
Is any of you capable of compiling and perhaps debugging or bisecting in which commit the problem appears?

ruevs, 2025-10-05:
Try with `-D ENABLE_OPENMP="OFF"`
https://github.com/solvespace/solvespace/issues/1579#issuecomment-2849396229

stesj4, 2025-10-05:
> I tried to compile it following https://github.com/solvespace/solvespace?tab=readme-ov-file#building-on-macos, but the cmake thing failed:
> 
> build % export LDFLAGS="-L/opt/homebrew/opt/libomp/lib"
> build % export CPPFLAGS="-I/opt/homebrew/opt/libomp/include"
> build % cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=ON
> -- The C compiler identification is AppleClang 17.0.0.17000319
> -- The CXX compiler identification is AppleClang 17.0.0.17000319
> -- The ASM compiler identification is AppleClang
> -- Found assembler: /usr/bin/cc
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: /usr/bin/cc - skipped
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: /usr/bin/c++ - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Error at /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:227 (message):
>   Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
> Call Stack (most recent call first):
>   /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:591 (_FPHSA_FAILURE_MESSAGE)
>   /opt/homebrew/share/cmake/Modules/FindOpenMP.cmake:652 (find_package_handle_standard_args)
>   CMakeLists.txt:136 (find_package)
> I updated and upgraded hombrew before. I am running on a M2 processor, could this be a problem?
> 
> My best guess for the problem would be an implementation issue in https://github.com/solvespace/solvespace/blob/master/src/platform/guimac.mm#L443. Maybe `NSTrackingEnabledDuringMouseDrag` needs to be added to the options (I just found it in https://stackoverflow.com/questions/6753727/mouseentered-event-disabled-when-mousedown-nsevents-mac?
> 
> Sorry if this is completely unrelated to the issue, I have never developed an AppKit software.

(I don't have Xcode,... installed so I can't test it)
if you just want to test and see if it solves the problem you should be able to run:
cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=OFF
you don't get the multithread support and the result is a slower program, but if it fix the drag problem there are tricks to get the multithread support
Good luck :)

stesj4, 2025-12-24:
I've got tired of this and tried FreeCAD. And this is what I found.

FreeCAD 1.0 absolutely terrible with more bugs than things that functions and a GUI that are nonfunctional. My guess it that it was build for Intel Macs and than just recompiled for silicon with very bad results, why else would so many use it on Windows and Linux with good result.

So I was not optimistic to try out 1.1 RC1. But it's absolutely great. A GUI that are great on Mac and I haven't found any bugs yet.

I also tried the 1.2 nightly build and it's better than 1.0 but quite many bugs that makes it difficult to work with.

So my journey with Solvespace end here. It was fun till it wasn't.

ruevs, 2025-12-25:
> So my journey with Solvespace end here. It was fun till it wasn't.

Enjoy FreeCAD.

As for this bug - obviously no one, that knows C++, uses SolveSpace and is willing to spend some time debugging this, is on macOS Tahoe... A symptom of a few things:

- Few people using open source mechanical CAD are programmers.
- Relatively few people use SolveSpace (compared to FreeCAD).
- Few (is it true?) programmers that find some bug in an open source program are willing to spend the time so set up the build environment and debug it.
- (relatively) Few people use macOS Tahoe.

It seems the intersection of the above four is currently the empty set :-) So this bug endures.

phkahler, 2025-12-28:
I merged the fix. If master works correctly we can close this.

closed 2025-12-31 by phkahler